EdgeTX/edgetx #5218: Radiomaster Zorro MPM Modul and Master/SBUS

arrenbrecht, 2024-06-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

With the Radiomaster Zorro, an external MPM module is not recognized and the MASTER/SBUS function via the module slot does not work either.
Up to EdgeTX version 2.8.4 everything worked perfectly.

### Expected Behavior

I wanted to use a wireless teacher/student system with an external MPM module (IRangeX lite) or a small receiver directly in the module slot via MASTER/SBUS. 
It works perfectly with EdgeTX version 2.8.4.

### Steps To Reproduce

Use an external MPM Modul or try to use the MASTER/SBUS Function via the  module slot.
With EdgeTX version 2.8.4 ist works fine

### Version

2.10.1

### Transmitter

RadioMaster Zorro

mha1, 2024-06-25:
which module bay pin did you use to connect the receivers SBUS output. Got to be SPort.

3djc, 2024-06-26:
I don't have a nano mpm, but I tested ELRS module ok

arrenbrecht, 2024-06-27:
I used the PXX_IN pin in the modul bay and it works with the EdgeTX version 2.8.4

mha1, 2024-06-27:
Use the SPort pin

arrenbrecht, 2024-06-29:
MASTER/SBUS with the SPort pin works.

But the MPM Modul does not.

mha1, 2024-06-29:
So two issues really.

External MPM: have you selected Multi in model settings / external RF?

arrenbrecht, 2024-06-30:
Sorry, MPM works, but without telemetry

mha1, 2024-06-30:
which protocol?

arrenbrecht, 2024-07-01:
Protocol is FrSkyX2, LBT(EU): no telemetry,
Protocol FrSkyRX: does not work.
tested in 2 different Radiomaster Zorro and in the Radiomaster Boxer.

in SYS/MODULES / RX VERSION:
Int. module
 Module      V1.3.3.20 AETR
Ext. module
 Module      No telemetry

3djc, 2024-07-01:
Wait, it means you have also an internal MPM on. Can you try turning it off ?

pfeerick, 2024-07-02:
> turning off the internal MPM makes no difference.

I can also at least get module information if both internal and external
modules are turned on on zorro running 2.10.1, using rm 4in1 module.

Have you verified that pins are undamaged on the external modules bay? And
just to double check, are you saying the external 4in1 was working with
2.8.4, but stopped when you upgraded to 2.10.1, with no other changes in
between? What 4in1 module is this with?

arrenbrecht, 2024-07-03:
I tested it again:
EdgeTX 2.8.4:
   selecting External HF-Modul 'MULTI'  -> Status V1.3.3.20 AETR
EdgeTX 2.9.0 or higher version:
   selecting External HF-Modul 'MULTI'  ->  Status no Telemetry

External Modul ist IRangeX IRX4 lite
tested in 2 different Radiomaster Zorro, one with internal MPM Modul, one with internal ELRS Modul an in a Radiomaster Boxer with internal ELRS Modul

3djc, 2024-07-03:
External module works absolutely fine with Boxer (and likely with Zorro too since you are the only one the report this issue), so I'm starting to think issue might be with your module

arrenbrecht, 2024-07-04:
I will close this issue now. 
However, I would like to point out that with EdgeTX version 2.8.4 the SBus input only functions on the PXX_IN pin and from EdgeTX version 2.9.0 the SBus input must be switched to the SPort pin if you want to use the MASTER/SBUS function.

closed 2024-07-04 by arrenbrecht

raphaelcoeffic, 2024-07-04:
> EdgeTX version 2.9.0 the SBus input must be switched to the SPort pin if you want to use the MASTER/SBUS function.

This has been fixed lately in `main` and `2.10` branches so that "Master/SBUS" input should be back in PXX_IN pin.