EdgeTX/edgetx #5231: DSMR Telemetry inconsistent refresh for telemetry and Spektrum DSMR Smart ESC RPM Value issue

jslatten, 2024-06-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I'm currently facing two issues with telemetry with my recently purchased Radiomaster MT12 4in1 transmitter and Spektrum SR315 DSMR receiver. I'm running the latest release of EdgeTX 2.10.1 and have updated the Multimod firmware  to V1.3.4.0 (TAER channel mapping if it matters).

The first issue is related to extremely inconsistent telemetry updates  from the connected Spektrum Firma Smart ESC (120a - 4s Max) when connected from my Radiomaster MT12 radio to the Spektrum SR315 Receiver. Sometimes I'll receive refreshed telemetry data every several seconds and other times I'll only get updated telemetry once every 1-2 minutes.  Sometimes I'll get no telemetry data at all which typically requires powering the vehicle and transmitter off and on several times until I eventually start to receive telemetry data from the Smart ESC again. I'm within 1-5 feet from the vehicle/receiver for all of these tests. I'm looking to receive consistently quick updates of the data for the following sources: Erpm, EVIN, TFET, ECUR, ETHR, and A2.  On the other hand I believe there are several other telemetry sources from the SR315 receiver or from the RM MT12 that seem to be consistently available and refresh quite often i.e TRSS, FDEa, FLss, Hold. I believe I'm receiving those every 1 second or possibly more often than that.  

I also have a Spektrum DX6C transmitter that has full telemetry and I've ran a number of tests with the Spektrum transmitter and it seems to have a live stream of all of the ESC telemetry that I'm interested in and it's being refreshed more than once per second it seems. Prior to switching to the RM MT12 transmitter I haven't experienced any issues with receiving telemetry on the Spektrum DX6C transmitter so I don't think it's the Spektrum receiver.

My second issue is related to the Erpm telemetry source that I'm receiving from the Spektrum Smart ESC...The data source is auto-detected with rpm unit type. I had noticed that the Erpm value was around double what the maximum should have been. Measured with Spektrum DX6C the RPM value was about 30500 rpm, but I was seeing values of close to 65,000+ rpm. I tried changing the number of poles to 4 to match the motor, but it was still incorrect. Afterwords, I changed the unit type from rpm to undefined and noticed the value reported was way too low so I tried different values for Ratio and Offset and ended up landing on around 13 for Ratio and 0 for offset. I think I'm in the general ballpark of where it should be, but may be slightly off with the ratio. I'd suspect the right Ratio value is in the range of 12.9 to 13.1. 

I'm at a loss as to why these issues are occurring, and I'm hoping that someone can assist in debugging this issue. 

### Expected Behavior

Receive nearly live updates of all Spektrum Smart ESC Telemetry i.e Erpm, ECUR, EVIN, etc

### Steps To Reproduce

1. Bind Radiomaster MT12 to Spektrum DSMR receiver using type DSM subtype R 1F.
2. Go to Model > Telemetry and Detect telemetry.
3. Monitor telemetry values coming from connected Spektrum Smart ESC i.e Erpm, EVIN, ECUR, ETHR, etc.

### Version

2.10.1

### Transmitter

RadioMaster MT12

### Operating System (OS)

Linux

### OS Version

Ubuntu 23.10

3djc, 2024-06-30:
@frankiearzu any thoughts ?

frankiearzu, 2024-06-30:
Don't have a surface RX to test, but with Avian ESC and Smart RX  (very similar in telemetry) refreshes properly Erpm, Evin,ECur, etc.
The EdgeTX Telemetry DSM code process the telemetry as soon as is passes from the Multi-Module. Maybe this could be a good question for the multi-module forum or github.

As for the Erpm sensor adjustment. 
1. The Unit RPM is calculated DisplayRPM = (SensorRPM * multiplier) / blades.   So for 1/2, set Multiplier=1, and Blades/Poles=2
2. If you use the generic unit (ratio/offset), ratio can be use to divide. Ratio=25.5 is 100%, so if you need 1/2, that will be a Ratio=12.7   (There is already an enhancement in the queue for that page to display percentage :-).   

Option 1 (Unit RPM) is preferable.

jslatten, 2024-07-03:
Thank you for the proper settings for RPM.  So for the inconsistent telemetry you are saying it's something with the multi-module firmware instead - I'll try posting something on their project as well.

frankiearzu, 2024-07-03:
Without knowing the formula is not that easy to decirpher what values to use. Had to read the code to remember :-)

For the telemetry refresh, I see my ESC data refreshed periodically without delayed.. but is an AVIAN ESC with an AR637 (For planes).    Don't have any surface RX to try to replicate.  Maybe Pascal (from Multi-module) can replicate it.

jslatten, 2024-07-03:
@frankiearzu just for curiosity sake when you say your telemetry is periodically updated around how often is that? Once per second? once every 5 seconds? I'm trying to determine what is normal to ensure my expectations are aligned. The other thing I was pondering is the Spectrum smart ESCs send telemetry data on the same channel as throttle instead of a separate connection to the receiver unlike most air receivers. I suspect the MM team will need to tell me whether that may be causing some conflicts with telemetry however it should be a common thing on surface Spektrum receivers.

frankiearzu, 2024-07-14:
Mine updates the AVIAN ESC telemetry at least once per second.. no delays that i can tell.

On the spektrum "Smart" Receivers, when the throttle (Port1)  is connected to an AVIAN ESC, the connection become digitial using SRXL2 protocol on the single signal wire.    From the TX->RX is passing all the channel positions, in this way the ESC can extract the throttle input as well as other channel when "reverse" is activated.      From RX->TX. telemetry data for the ESC and battery is sent.        

Try connecting the Throotle to the Bind port instead.. the Bind port is a 2nd  SRXL2 carrying the same data. I know, seems strange, but worked.. im using this to drive an EDF afterburner LED: AVIAN connected to the Bind (uses SRXL2), and afterburner LED on Throttle/Port1 (analog). Both reponding to the Throttle channel.

Just FYI, when the Throttle/Port1 is NOT connected to an AVIAN ESC, it acts as a regular servo pulse outout (analog, no telemetry).

jslatten, 2024-10-19:
Thanks @frankiearzu...I honestly think this telemetry issue is more so an issue with being too close to the receiver. If I'm farther away it seems to work okay. Although It still doesn't update as fast as my spektrum transmitter, but it's usable. Now the more annoying issue is the range of both telemetry and controls - I can't use it too close or too far away. lol. It seems the 4in1 is significantly shorter range. I suspect it could be something with the filtering or selectivity of the transmitter and receiver electronics. None of these are EdgeTX issues so I guess I'll close this one out and keep checking back with the MM team from time to time.

closed 2024-10-19 by jslatten