EdgeTX/edgetx #5234: ISRM mit X9 Lite S

Fredflar, 2024-06-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Sender lässt sich nicht auf Modul ISRM einstellen.
Sender cannot be adjusted to Module ISRM.

### Expected Behavior

Beim wechsel von OTX zu EdgeTX kein Ausfall des Moduls. Bei meiner X9D+2019 ohne Fehler des Moduls.
When changing from OTX to EdgetX, no failure of the module. With my X9D+2019 without mistakes in the module.

### Steps To Reproduce

Mehrmals das Modell, die Hartware Einstellung geändert kein Erfolg. Radio Firmware.bin neu aufgespielt und Bootloder aktualisiert ohne Änderung. Neu aus Internet heruntergeladen.
Several times the model, the hardware setting changed no success. Radio firmware.bin re -played up and updated boot looder without a change. New down from the Internet.

### Version

2.10.1

### Transmitter

FrSky X9 Lite / Lite S

### Operating System (OS)

Windows

### Anything else?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

pfeerick, 2024-06-30:
It appears there is a bug where you can only select between CRSF and MULTI if ISRM is not already set. 

Edit your `RADIO/radio.yml` file (on the radio's SD card) and change `internalModule` to `TYPE_ISRM_PXX2` and you should be able to select ISRM in your model settings now (you will need to toggle it off and on again if internal RF was turned on).

raphaelcoeffic, 2024-07-01:
> you can only select between CRSF and MULTI if ISRM is not already set

I just verified: by default, ISRM is selected (starting with empty SD card). However, as soon as something has been selected, no way to go back to ISRM.

closed 2024-07-01 by pfeerick

Fredflar, 2024-07-01:
Danke

PR #5236 (2024-07-01): fix: supported internal module for monochrome radios w/ ISRM
Fixes #5234
Fixes #5153

pfeerick: I am going to tentatively say this resolves #1942 also as this gem has been present since 2.6, so could have been responsible since it was a 2.7 issue. 

Summary of changes:
- use `DEFAULT_INTERNAL_MODULE == ISRM_PXX2` as a better heuristic to determine the supported modules.

Note: it seems X9lite and X9liteS are impacted as well.