EdgeTX/edgetx #5246: Custom screen widget doesnt work anymore and latency on Manage Models page.

Kevltan, 2024-07-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I load recent nightly firmware (I can only go back 2 weeks), my screen widget no longer functions correctly. I initially get all the trim functions laid over the my widget screen, which are disabled in screen settings. Then when touching the screen I lose them and end up the EdgeTX Logo in the top left corner of the screen. I can no longer access widget options by tapping the screen as before. 
Also now I have loads of latency when accessing Manage Models, instead of the model windows coming on to screen all at once they cascade on, one at a time every about 0.2 of a second. Returning to an earlier 2.11.0 nightly removes issue.

### Expected Behavior

No sliders or EdgeTX logo on should be overlaid on screen widget when disabled. Widget "options" to be accessable by touching widget screen.

### Steps To Reproduce

When loading all "available" latest nightly firmwares. Have tried latest updated SD card too.

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12

### Anything else?

Latest downloadable nightlies

pfeerick, 2024-07-04:
Model images loading "on demand" while the system is idle rather than
slowing the load of the screen is expected behaviour. It is not latency -
you will be able to interact with the screen rather than it be frozen or
having had to wait for those images to load before being shown to you.

philmoz, 2024-07-04:
The EdgeTX logo appearing when you turn off all the decorations (trims, sliders, flight mode etc) is by design.
See issue #4371 

The trim displays should disappear when you turn off the trim decoration in the screen setup. This works correctly on my radios. If this is not happening please post a video or photos of the behaviour.

pfeerick, 2024-07-04:
Ah, this makes sense now

>  Double tapping the screen removes logo but I can no longer access widget "options" by tapping the screen as before.

Yes, because you have now made it so the widget gets all key and touch input now, not the system. You would need to long press RTN as mentioned in #4655, and then you should get options again on long press. 

I think we forgot about how to bring up options for the widget here with full screen layout though? Since Long enter no longer goes into "select a widget to interact with" mode so you can bring up the widget options, and a long press toggles "App mode" / full screen.

As an extension of #4655... and also completing some of the TODOs... rather than 'changing' "Full screen" to "App mode", why not add Add mode as it's own option?

i.e. put a toggle switch *above* the layout option for app mode. When you toggle that on, it can disable / hide all the layout controls, auto-magically set them to off, as well as layout to full screen, leaving just the setup widget button. With it being it's own screen option, you should be able to detect whether it is on or off, when determine how to respond to user input at the main views then? i.e. whether to switch to app mode, or show options?

Kevltan, 2024-07-05:
Before latest nightly

https://github.com/EdgeTX/edgetx/assets/56512603/87dc1a23-dbba-4ff0-9f7c-f0fd0cfeba00

https://github.com/EdgeTX/edgetx/assets/56512603/c91646de-c114-400f-9377-eef9977d8637

Latest nightly

https://github.com/EdgeTX/edgetx/assets/56512603/dbe0826e-4f90-48a9-986b-e1b04586cd81

https://github.com/EdgeTX/edgetx/assets/56512603/63963535-e485-40c5-8b29-e696213091a2

pfeerick, 2024-07-05:
With respect to the model select screen only:

And what happens the **second** time you open the screen after having turned the radio on? Does it load "instantly" like in the "before" video? As I think for your *before* video, you had previously had the model select screen open, thus the images were cached in memory. Since normally, the first time you open manage models after booting the radio, the radio would have been unresponsive from when you long pressed the MDL button, for the time it took for all those images to load... i.e. if it is 0.2 second per image , and 10 images... it would previously have taken nearly 2 seconds before the screen even came up. Versus coming up immediately, and progressively loading the images as it does now.

philmoz, 2024-07-05:
> With respect to the model select screen only:
> 
> And what happens the **second** time you open the screen after having turned the radio on? Does it load "instantly" like in the "before" video? As I think for your _before_ video, you had previously had the model select screen open, thus the images were cached in memory. Since normally, the first time you open manage models after booting the radio, the radio would have been unresponsive from when you long pressed the MDL button, for the time it took for all those images to load... i.e. if it is 0.2 second per image , and 10 images... it would previously have taken nearly 2 seconds before the manage models screen even came up. Versus coming up immediately, and progressively loading the images as it does now.

The image cache is currently deleted when you leave the manage models page - it's there to avoid reloading images when you change labels.

Kevltan, 2024-07-05:
All were done from power up there is no sequence event  on my original firmware - manage models  just all come on as a block.

philmoz, 2024-07-05:
The "App Mode" behaviour is described in issue #4371 
You should read the discussion there regarding how it works.

It is implemented how it was requested - specifically this from @JimB40 
[image]

pfeerick, 2024-07-05:
Ok, so I need to have words with Robert, as he is 100% wrong on both of those points. :facepalm: 

Then again, implementing app mode properly - i.e. it being *separate* to fullscreen as I proposed above, would resolve all of this. Since "App mode" != fullscreen widget. Full screen widgets **still** have options, and you need to be able to access them!

Kevltan, 2024-07-05:
Looks like Im stuck in the past  and anybody else who uses TD-RDT as this change does not complement this lua script. This means that I cannot move forward with firmware updates anymore. 😒

Kevltan, 2024-07-05:
Yes in my original firmware 2.11.0 the model page comes on instantly - no delay with images loading in sequence.

Kevltan, 2024-07-05:
fix(color): restore previous 'Fullscreen' functionality and add new 'App mode' layout #5248
🙂🙂👍👍

Kevltan, 2024-07-05:
https://github.com/philmoz
 Have tried your PR but still does not work correctly as trims and and logo are still present when powering up and when switching to another model and back again. I have to touch screen and hold for option box press full screen and then hold return key to enable widget settings again. This is in full screen mode.

Kevltan, 2024-07-05:
@philmoz
@pfeerick 
Yep its Fixed! Superb and excelent turn around as usual 😊👌👍👍

PR #5248 (2024-07-05): fix(color): restore previous 'Fullscreen' functionality and add new 'App mode' layout
Fixes #5246 

PR #4655 adds the requested 'App mode' functionality; but breaks existing 'Full screen' functionality for some widgets.

This PR restores the old functionality when the layout is set to 'Full screen'.
A new layout called 'App mode' implements the new app mode functionality.
The 'App mode' layout has a single widget and disables topbar, sliders, trims and flight modes.

[image]

[image]
closed 2024-07-05 by pfeerick