EdgeTX/edgetx #5249: Generated Firmware.bin is not a valid

atrbx5, 2024-07-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

cmake -DPCB=X12S  \
   -DCMAKE_PREFIX_PATH=$QTDIR \
   -DARM_TOOLCHAIN_DIR=/Applications/ARM/bin/ \
   -DPYTHON_EXECUTABLE=$(brew --prefix)/bin/python3 ..

make -j8 firmware   

this should generate valid firmware for tx12 mk||

### Expected Behavior

Companion and transmiter both said that compiled firmware.bin is not valid firmware

### Steps To Reproduce

cmake -DPCB=X12S  \
   -DCMAKE_PREFIX_PATH=$QTDIR \
   -DARM_TOOLCHAIN_DIR=/Applications/ARM/bin/ \
   -DPYTHON_EXECUTABLE=$(brew --prefix)/bin/python3 ..

make -j8 firmware   

firmware.bin with 1.5mb size

### Version

2.10.2

### Transmitter

RadioMaster TX12 / TX12MK2

### Operating System (OS)

macOS

philmoz, 2024-07-05:
For TX12 you should use -DPCB=X7 -DPCDBREV=TX12
For TX12 Mk2 you should use -DPCB=X7 -DPCDBREV=TX12MK2

pfeerick, 2024-07-05:
Also, look at https://github.com/EdgeTX/edgetx/blob/main/tools/build-gh.sh when trying to figure what flags to use for which radio ;) By specifying `-DPCB=X12S`, you're building firmware for the Frsky X12S, not the Radiomaster TX12 mk2 that you intended to build for... hence the bootloader is correctly saying that firmware isn't right for it ;)

closed 2024-07-05 by pfeerick

atrbx5, 2024-07-05:
Thanks i did look what are the options and thought i understand it (nope)
did compile and flash but only from sd card. After flashing with EdgeTx companion TX not start