EdgeTX/edgetx #5250: Posssible bootloader 2.10.x issue on FrSky X10(S) Express radio.

Merga189, 2024-07-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

I see a possible issue on my FrSky X10S Express radio when loading firmware via the Edge bootloader.

Started a discussing in  https://www.rc-network.de/threads/edgetx.11851434/page-57#post-12801905

First thought, a problem with an issue on my side. Faulty SD card, as I am on Linux here, but building a new SD card under Windows 10 (formatting in FAT32, loading the FW files fresh from here https://github.com/EdgeTX/edgetx/releases/ showed the same behavior. The bootloader, which ship with 2.10.0, 2.10.1 and 2.10.2 don't accept the  FW files. but NO PROBLEM when reverting to the bootloader release 2.9.4!

### Expected Behavior

Working bootloader in releases 2.10.0 and later ....

### Steps To Reproduce

Loading the FW via bootloader on FrSky(S) Express. 

### Version

2.10.2

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Linux

pfeerick, 2024-07-05:
Just reading through some of the comments from that thread
- no, it is not a Linux / OS issue... as I for one most certainly would have experienced it as I reguarly switch between Windows and Linux. 
- this could happen if the wrong bootloader were been flashed to the handset (and by luck works)... as 2.10 bootloader checks if the firmware's radio type matches that of the radio type that is set in the bootloader. i.e. if you were to flash an x10s bootloader , and attempt to flash x10s-access firmware, it would show that exact message

I wonder if the name/type is a problem here... as it is more than 8 characters?

Merga189, 2024-07-05:
You did see the screenshots in that thread on RCN!?

I do have the FW files with the full! name on the SD card and here user Voodoo-68 has it on his X10 with full name
https://www.rc-network.de/threads/edgetx.11851434/page-58#post-12802446 without that issue.

I shorten the file names for a test and come back here .....

Merga189, 2024-07-05:
Short test. renamed (shortened) the 4 FW files names on the test SD card

[image]

Same behavior 

[image]

pfeerick, 2024-07-05:
No, not the file name. I'm thinking the internal test may be having issues
as it says the firmware type is x10-acce (the last two letters - ss - being
missing). Hence why voodoo-68 would not be having issues with X10. If I'm
right, this would affect one, maybe two other radio variants.

3djc, 2024-07-05:
- What's that x10-access ? It should be x10express (flavor is used)

https://github.com/EdgeTX/edgetx/blob/6f8a5313d5341116a020f0b3ebe01393c2671ead/radio/src/targets/horus/CMakeLists.txt#L67

pfeerick, 2024-07-05:
I'm on mobile, so it could be express. Regardless, the last two letters are
missing in the photos, and x10express is longer than 8 characters, if that
means anything here...

Merga189, 2024-07-06:
Can I test that by renaming the FW files on 2.10.x ??

gagarinlg, 2024-07-06:
No, it is not the file name, but the radio name stored inside the file.

pfeerick, 2024-07-06:
@Merga189 Can you test the PR fix? I've attached the build of that for the X10 Express based on 2.10, so it won't hurt any settings if you do flash both the bootloader and firmware... 

[x10-access-pr5255.zip](https://github.com/user-attachments/files/16115540/x10-access-pr5255.zip)

Merga189, 2024-07-06:
[image]

[image]

Any further tests necessary? Looks good! Thank you ...

Karl

Merga189, 2024-07-06:
Maybe you want to see this pic also

[image]

3djc, 2024-07-06:
All good, thanks

Merga189, 2024-07-06:
I think, I'll close this now.

Karl

PR #5255 (2024-07-06): fix: VersionTag.flavour[] too small for x10express
Fixes #5250
 Also prevent future occurrences by checking size at compile time
closed 2024-07-06 by Merga189