EdgeTX/edgetx #5251: 2.10.2 Tx16s flash from bootloader stuck @ 90%

Ga1j1n, 2024-07-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Updated from 2.10.1 to 2.10.2 via the bootloader flasher (also updated first along with SD) it seems the flash has stuck at approx 90%, how long do I wait?, normally it's very quick, Suspect I'll brick it if I interrupt

### Expected Behavior

Standard flash behaviour, flashed this way many times

### Steps To Reproduce

1) left right trims in and power on to enter bootloader 2) select firmware file 3) select flash 4) lockup

### Version

2.10.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Not applicable

Ga1j1n, 2024-07-05:
recovered using STM32Cube Programmer, hopefully isolated
https://www.youtube.com/watch?v=H9_sxXY_K-8

Merga189, 2024-07-05:
Interesting. Which bootloader version did you use? I see here an issue with the bootloader version 2.10.x on my X10S express.  

https://github.com/EdgeTX/edgetx/issues/5250#

Reverting to bootloader 2.9.4 and it works fine here.

br Karl

pfeerick, 2024-07-05:
This is not new to 2.10, but has been happening every so. It is recoverable
on handset simply by removing the battery, reconnecting it, and doing the
firmware update again. You can't brick the radio by interrupting, as only
the firmware is being written, not the bootloader.

Ga1j1n, 2024-07-06:
This was with bootloader 2.10.2

3djc, 2024-07-06:
It is a random event, already existed in otx

dirteat, 2024-10-30:
Sorry to add to the fest that updating from 2.9.0 to 2.10.5 today, did the same for me. Bootloader writing failed around 80% forever. And yes, after a while I removed the battery and it is bricked :(

pfeerick, 2024-10-30:
This comment still applies. It will not be bricked... you should still be able to re-enter the bootloader, just not the main firmware since that obviously did not finish writing. 
 https://github.com/EdgeTX/edgetx/issues/5251#issuecomment-2211392754

dirteat, 2024-10-30:
Maybe I misunderstood how to enter the bootloader, but it is not booting anymore, cannot start again the firmware update even pressing T1/T4 + power. Nothing... And I've been spending 3 hours now trying to download the superb STM32CubeProgrammer on that very special website...

pfeerick, 2024-10-30:
No, that is the correct way... Ok, something really strange going on there. Did you update the bootloader before trying to update the firmware?

Do you need to download STM32CubeProgrammer (i.e. do you need it for the driver? If you are running a chrome based web browser, you may be able to do the firmware update online, unless you are unlucky enough that one of the fuse settings on the MCU needs changing. 

And yes, the STMicro website can sometimes be... Unhelpful... there has been times when it will not let me log in... (making that a job for another day) and others where the download page refuses to give me the link... I just refresh it every few 5-10 minutes,  give it plenty of time to think and load, and it usually finally coughes it up.

dirteat, 2024-10-30:
Well, I've followed exactly the update instruction with the bootloader method:

https://manual.edgetx.org/installing-and-updating-edgetx/update-from-an-earlier-version-of-edgetx-using-the-bootloader

So, yes, I've first update the bootloader (navigating the menu etc), which went well, then went on rebooting the radio with T1/T4 pressed, and selected Write Firmware and it hanged out there. Then, I've removed the battery after a long while.

The only things I was hesitating too was to either put in FIRMWARE tx16s-6b539d0.bin (from edgeTX repo) or the tx16s-EN-bluetooth-9309ea2.bin, and I did choose the second one. May that was my mistake?

Anyway, I've got the linux version of STM32 (I've spend some time trying to use it under windows7, it is not supported anymore...). Time to play :)

dirteat, 2024-10-30:
Woo, once you get that prog, that is fast. I've flashed tx16s-6b539d0.bin this time, and job done. I'll try to redo the exact procedure I've done to see if I can reproduce the bug. I have another TX16II I need to update (from 2.7 this time, so I could also test this one).

pfeerick, 2024-10-30:
Fantastic! Not entirely sure what the problem is... as JC alluded to above, it is nothing new .. as it did fail in the OTX days also .. but usually a battery pull is all that is needed. Possibly opening the TX up and removing the RTC battery also _might_ have unstuck this. 

It's going to be damn near impossible to debug as it only happens every so many flashes (like 1 in 30, and then it's *maybe* it will happen 🤪), and you'd need to be lucky enough to have the hardware debugger connected when it failed... and even then it may not have anything useful to tell you :(

dirteat, 2024-10-30:
Thanks for the feedback and the help. I guess the most difficult part is to actually download STM32CubeProg :-)

pfeerick, 2024-10-30:
No problem, glad you got it up and running again. And yeah, it is a pain to
download at times, but a great tool once you do have it.

LevYas, 2024-11-23:
I've just had this problem when updating 2.8 -> 2.10 and this thread helped. I disconnected the battery, in 10 seconds connected it again and loaded in T1+T4+Power mode, bootloader started and I initiated writing again, this time it went well.
Should I make a PR to add this trick to that page? https://manual.edgetx.org/installing-and-updating-edgetx/having-trouble-updating

pfeerick, 2025-02-07:
@LevYas Yes please, PRs (especially to documentation! 🤭) are always welcome. It really does need to be written down somewhere so can just point people to that snippet. Since this really is a random (and somewhat known) issue, and can be resolved with hard power cycle or DFU flashing, going to close.

closed 2025-02-07 by pfeerick

LevYas, 2025-06-02:
Done! https://github.com/EdgeTX/edgetx-user-manual/pull/53