EdgeTX/edgetx #5259: Pr 5228 corrupts sd card

tonims1, 2024-07-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Since the pr 5228 has been merged with my tx16 mkII, selecting a model corrupts the SD card.  You can't read any files from the PC either.

### Expected Behavior

that the sd card is not corrupted

### Steps To Reproduce

When you want to select another model there are already images that do not appear.  Selecting one corrupts the SD card.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

windows 11

3djc, 2024-07-07:
Could you test with a firmware in https://github.com/EdgeTX/edgetx/actions/runs/9825787543/artifacts/1674748063 ?

tonims1, 2024-07-07:
Now it seems that everything is fine.
I send the configuration file before and after. without making any changes to changed something. I guess it's normal
[5tmp.zip](https://github.com/user-attachments/files/16118633/5tmp.zip)

pfeerick, 2024-07-07:
By send, do you mean you are using writing settings with Companion? If so, yes, the radio firmware does rewrite/reorder stuff to suit itself if the Companion is not in the same order as the radio wants it. As long as stuff isn't getting corrupted... i.e. you can edit it on the radio, make changes, restart it multiple times, and the radio is now working... fingers crossed it's all good.

tonims1, 2024-07-07:
ok Thanks

pfeerick, 2024-07-07:
Please leave the issue open. It will be automatically closed if/when a linked PR is merged. Also, the issue is not actually resolved until that happens.

tonims1, 2024-07-08:
OK thanks

Kevltan, 2024-07-09:
@pfeerick 
This works on my TX16S no corruption of SD card!

closed 2024-07-10 by pfeerick

Kevltan, 2024-07-10:
@pfeerick 
All working fine again including IMU. Thanks Peter👌😊

pfeerick, 2024-07-13:
NP. Credit goes to Raphael for the fix thou ;)

Kevltan, 2024-07-14:
Hi Peter,  I left a note on Discord about the issue I am having with the
radio switching off  indescriminately when the USB cable is plugged into
TX16S. I think there is a conflict between the USB and IMU firmware as this
doesn't happen when I remove the cmake addition for IMU in the build. Any
ideas?

PR #5260 (2024-07-10): fix: SDIO issues after H7 support
Fixes #5259

Summary of changes:
- reinstate GPIO init as they were before (pull-ups, speed very high)
- better DMA buffer check