EdgeTX/edgetx #5261: x-lite internal IXJT-XLITE not accessible

chrisu365, 2024-07-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The radio is an x-lite accst model using  an internal IXJT module flashed by IXJT-XLITE_ACCST_2.1.0_LBT  working fine on Open TX 2.3.15
After flashing Edge TX v2.10.2  selecting xlite-1de0600.bin  the internal RF can only choose between MULTI or CRSF
EdgeTX v2.10.2 is unable to flash the internal iXJT RF module of the Frsky X-Lite
The bug was already fixed for radios using an ISRM module:
https://github.com/EdgeTX/edgetx/issues/1552
Any chance to get a fix for radios with internal IXJT module ?

### Expected Behavior

Internal IXJT should be usable

### Steps To Reproduce

    Firmware is stored on SD card
    Select file and option to "Flash Int. Module"
    "Device reset..." is displayed and within 5 seconds beeps
    Following beep you get error "FW update error" "Device not responding"
    Reverting to OpenTX and flashing process works as expected
    Video of process and failure available on request.

### Version

2.10.2

### Transmitter

FrSky X-Lite / S / Pro

pfeerick, 2024-07-07:
For the moment, just to get it working again, you can edit your `RADIO/radio.yml`, and change `internalModule` to `TYPE_XJT_PXX1` . Make sure you also change `manuallyEdited` up the top from `0` to `1`, so that the radio accepts the invalid file checksum that will result from the edit to the file ;)

```yml
internalModule: TYPE_XJT_PXX1
```

chrisu365, 2024-07-07:
Compiled a firmware using https://github.com/EdgeTX/edgetx/pull/5262
The internal XJT module works 👍

pfeerick, 2024-07-07:
Fantastic! Thanks for the confirm that it works as expected :)

PR #5262 (2024-07-08): fix: configure default internal module for XJT_PXX1
Fixes #5261

Just double checking INTERNAL_MODULE_SERIAL _is_ correct here?
closed 2024-07-08 by pfeerick