EdgeTX/edgetx #5274: Option for not saving trims

3djc, 2024-07-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

A

### Describe the solution you'd like

B

### Describe alternatives you've considered

C

pfeerick, 2024-07-09:
IMO, 

> an option for not saving trims

would be the best, since UI wise it can go in the already existing trim section on colorlcd - and on B&W there would be  enough trim options with this now to warrant a trim heading and collapsible section. Having it as a trim reset SF seems like a bit of a cludge... but that itself could be another option :zany_face: Unless you think there could be a need/want for this to be configurable on a per-mode basis.

philmoz, 2024-07-09:
Wouldn't that also require sending trims to the receiver as a separate data element?
If so would it also require changes to the mixer to remove trim calculations.

pfeerick, 2024-07-10:
Not necessarily. If the receiver is saving the last state, we simply need the the TX (edgeTX) to not save the trim state / reset back to zero. It really depends on whether the receiver considers the trims as a separate construct or not. I guess I'm saying we really need more information on the behaviour of this protocol / RX-TX pair before can fully understand what is going on. And/or how the MPM is translating things ;)

philmoz, 2024-07-10:
How can the receiver save just the trim state if it does not get the trim data separately?

pfeerick, 2024-07-10:
That is a big assumption ;) Depending on the operation of the unit, *it may not get it* separately on power-down or after a time period. Some TXs also only transmit changes, or even only on button presses. There are also protocols like the some of the E129 subprotocols which use separate channels for trims (or at least, to communicate the trims to the MPM) - which also remember the offset. The world of the MPM and all the various protocols is a weird and wonderful place ;)

3djc, 2024-07-10:
Just had a lengthy conversation with Pascal, who was kind enough to educate me on those models.

Stock radio:
At each lipo change, the transmitter needs to be turned off
Upon start, it will initiate binds, and start with trims at 0
During operation, the receiver stores continuously the last trims values (hint)
Yes, trims values, those are sent using separately from stick values (btw, there is often no more than 60 possible stick values on those (!))
Currently, for lack of better information, we assume the radio/receiver works that way so it can be used on several models without the need to support models. Pascal mentions it is becoming more and more mainstream in the 'toy model' category

EdgeTX
Model setup:
- you need a switch to activate bind on channel
- you need to deactivate trims on your aetr channels
- you need to send trims values as separate channels
[image]

Operation
At each lipo change,  you need to trigger a bind on channel from a switch (already supported), Pascal usually resets timers too
Thats where trim reset also need to take place. Not saving trim is not an option, since it would need model reload/transmiter reboot to get zero trim, so what is really needed is a reset trims feature (part of the reset SF ?)

Clever operation
As shown in the screenshot above, to go further than the allowed flight enveloppe defined by manufacturer, Pascal does temporarily add sticks to trims to get full stick value + full trim value

pfeerick, 2024-07-10:
Sounds like it behaves just like the E129, with CH7, CH8 and CH9 being the trim channels. I agree, trim reset as another sub-option of the reset SF is probably the best option... certainly would want to need to turn the transmitter on/off on a battery change if going through multiple batteries in a session...

ParkerEde, 2024-07-11:
This is also a good feature for any protocolls if you are working with a stabilizer that takes over the current trim by switching a channel three times within 1 second. The trim is then saved in the stabilizer and EdgeTx would no longer have to remember this trim and could jump back to 0. Right?

3djc, 2024-07-11:
Never worked with such a stabiliser, but it sounds like it could be used there too

PR #5278 (2024-07-28): feat: add "Trims" option to Reset special function 
Fixes #5274

Summary of changes:
adds "trims" choice to reset function. Activate reset all trims to 0 in current flight mode.
closed 2024-07-28 by pfeerick