EdgeTX/edgetx #5275: R9SX receiver with R9M module, Horus x10s

jetfanZK, 2024-07-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

To be honest - I'm not sure if this is a bug:

[image]
[image]
[image]
[image]
[image]

I can't bind R9SX receiver with R9M module in Horus x10s transmitter with IRSM module and Edge 'Centurion' 2.10.1
R9M module and R9SX receiver are after flashing in LBT ACCESS version. I try to bind the receiver: select 'Bind' in the module setup and power on the R9SX receiver with the F/S button pressed. I release the button, turn receiver off and on and... nothing happens. I tried many combinations - no results. What am I doing wrong? I have not found solution on the web, can you help me please?

### Expected Behavior

Binding with receiver

### Steps To Reproduce

Described above

### Version

2.10.1

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

3djc, 2024-07-10:
This module is not a R9M, it is a R9M 2019 module. R9M 2019 is only offered as a choice on X10Express radio (if I remember correctly, X10 hardware doesn't support it (but I'm not fully sure))

[image]

pfeerick, 2024-07-10:
Simulator suggests not... not presented as an option there. 

Looks like it would need the [externalaccessmod](https://github.com/opentx/opentx/wiki/Externalaccessmod#frsky-x10) done, then it would be possible to enable it by setting AUX1 in Radio Settings -> Hardware to External Module, o

closed 2024-07-10 by pfeerick