EdgeTX/edgetx #5285: Horus X10S Express: MPM switching off telemetry no longer works

ParkerEde, 2024-07-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I am using a Horus X10S Express with an external MPM 4in1 irange+ with a Blade130S. Telemetry can be switched off in the DSM protocol options. Since 2.10.? you can set the switch to deactivated, but the telemetry remains active. However, since there is a problem with telemetry in that the serial connection to the MPM is sporadically lost when telemetry is active, the last solution is to fly a Blade130S or 230S in combination with the Horus X10S Express and MPM. See: https://github.com/EdgeTX/edgetx/issues/4411

### Expected Behavior

Telemetry should be able to be switched off again

### Steps To Reproduce

Bind with DSM RX with telemetry, open DSM Option in MPM, deaktivate Telemetry, observe there is still telemetry activ.

### Version

2.10.2

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

3djc, 2024-07-11:
Could you test with this 
[x10e-test.zip](https://github.com/user-attachments/files/16174600/x10e-test.zip)

ParkerEde, 2024-07-11:
@3djc  
that works.

3djc, 2024-07-11:
> @3djc that works.

Cool, I have made a 2.10 based version if you want to use it (previous was based on main, so inherently less stable)
[x10e-test.zip](https://github.com/user-attachments/files/16175439/x10e-test.zip)

ParkerEde, 2024-07-11:
wow, great. Thx. Could we have this fix official in 2.10.3?

PR #5286 (2024-07-13): fix: MPM disable telemetry user choice ignored
Fixes #5285

Summary of changes: overrides auto detection with user choice for MPM telemetry disabling
closed 2024-07-13 by pfeerick