EdgeTX/edgetx #5291: Trims still visible after being set to off in model setup page

alant101, 2024-07-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Unable to remove trims from display

### Expected Behavior

Not to be visible

### Steps To Reproduce

select option for trims not to be displayed

### Version

2.10.2

### Transmitter

FrSky X9D+2019

3djc, 2024-07-12:
What option did you exactly use ? I think you misunderstood what it does, since there is none in model settings to do that

alant101, 2024-07-12:
What I was hoping to achieve is to remove the trims for the display 
[X9D 2019.zip](https://github.com/user-attachments/files/16196498/X9D.2019.zip)

3djc, 2024-07-12:
Trims will be removed from display only of deactivated in flight mode

alant101, 2024-07-12:
Okay thank

pfeerick, 2024-07-13:
> I think you misunderstood what it does, since there is none in model settings to do that. The closest is trim values.

That is easily done, since the text is "Display Trims", not "Display Trim **Value**".

alant101, 2024-07-14:
Hi, what I'm trying to achieve is, remove trims from display. As I wish to
use the Inav lua script and hence no trims visable. If you're able to help
me with that, I would be very grateful

3djc, 2024-07-14:
Unrelated, the lua script run out an empty screen anyway

alant101, 2024-07-15:
> Even so, JC has already told you how to turn the trim display off... By
> going into flight/drive modes, and actually disabling/turning off the
> trims
> not used. Then they won't be displayed on the main views either.

Just like to say thank you, I now understand what you were referring to

3djc, 2024-07-15:
But again, that does not affect AT ALL inav lua scripts

closed 2024-07-15 by pfeerick