EdgeTX/edgetx #5292: Lua:  getFieldInfo using "Min" or "Max" always return min.

frankiearzu, 2024-07-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

While using the Flights Widget, noticed that regardless if i choose MIN or MAX for the source, it reports "min" for MAX, and "--" for min.

### Expected Behavior

Should properly report the source name.

### Steps To Reproduce

1. Setup the Flights widget.
2. Choose MIN as the source for the ARM
3. go to full screen to see the name of the source/switch

### Version

2.10.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

[image]
[image]

frankiearzu, 2024-07-13:
The problem is here:   Mistyped constant...  (Ill do the PR to fix it).
[image]

mha1, 2024-07-13:
I blame copy/paste. Thanks for making this right. And obviously not the most used LUA constant. No complaints since 08/23.

PR #5293 (2024-07-13): fix(lua): incorrect name for MIN source
Fixes #5292

Summary of changes:
Corrected the source constant to use for min,

Now it returns properly min or max.
closed 2024-07-13 by pfeerick