EdgeTX/edgetx #5296: Higher precision delay up/dn

jonathanthompson-me, 2024-07-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The problem is certain types of racing need higher precision with delay up/dn. This is related to issue #4593 . 

### Describe the solution you'd like

Same solution as issue #4593 , but for delay up/dn.

### Describe alternatives you've considered

Global variables of drive modes could be used if math was able to be applied to them to get a higher precision value to be used for the delays. I haven't figured out how to multiply a percentage variable by another variable. It seems the global variables are just meant for simple comparison of value.

### Additional context

This is mainly for very fine grained throttle control with the Racemaster MT12 when drag racing, but might also be valuable for other types of racing.

jonathanthompson-me, 2024-07-17:
We are gonna want the 0.0x or better precision on logical switch duration and delays too. In order to build the tools for drag racing, with the precision that will be required to be competitive, it's gonna be needed.

arhifisyant, 2024-07-17:
yes, agreed. I face the same wall barrier too in logical switches during programing my car. Im using zorro btw. So, just implement this on all tx model. it will fine i think. Please make this happen. We need 0.0x precision in logical switches (delay and duration)

philmoz, 2024-07-17:
I think you mean 0.00 precision (1/100th second). Mix delay and LS delay and duration are already 1/10th second (0.0 precision).

Mix delay could be done.

Logical switches would be problematic as they are currently only updated every 1/10th second.
To implement 1/100th second precision would require running the logical switch processing code every 1/100th second.
This may have other side effects and could affect performance.

arhifisyant, 2024-07-18:
Oh, very well explained. Its about looping frequency in the LS area. Slightly out of topic, can we actually monitor the CPU/mem usage in the transmitter, realtime?

jonathanthompson-me, 2024-07-18:
> I think you mean 0.00 precision (1/100th second). Mix delay and LS delay and duration are already 1/10th second (0.0 precision).
>
> Mix delay could be done.
>
> Logical switches would be problematic as [...]

That's great info to have but will force me to rethink my implementation. I'm still interested in the mix delay to the hundredth.

PR #5314 (2024-07-29): feat: add precision setting for mix delay up & down
Fixes #5296
closed 2024-07-29 by pfeerick