EdgeTX/edgetx #5297: Edgtetx 2.10.1 9XE forgot the installed sensor

MicheleVilla, 2024-07-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In telemetry page search and visualize the installed telemetry sensor (sensor cels, altimeter GPS etc)
When switch off and switch on the transmitter the sensor in subject are missing and Is necessary to make another search sensor operation.

### Expected Behavior

Once the sensors installed on the receiver have been searched for and memorized, they should remain available even after restarting the transmitter.

### Steps To Reproduce

Go in telemetry page
search the installed sensors
switch off the transmitter
switch on trasmitter and go in telemetry page
sensors are missing

### Version

2.10.1

### Transmitter

FrSky X9E / X9E Hall

3djc, 2024-07-15:
Have you simply not toggled the arrow?

MicheleVilla, 2024-07-15:
Oppssss....
You are right
I think this issue can be closed

closed 2024-07-15 by MicheleVilla