EdgeTX/edgetx #5299: Slow, choppy playback of the model name file.

Plohish67, 2024-07-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If I try to switch from another model, does the phrase "simulator" sound stuttering.
If the model is already selected when the equipment is turned on, the phrase “simulator” is reproduced perfectly! 

### Expected Behavior

Smooth, normal playback of the model name file.

### Steps To Reproduce

1. Open the model manager.
2. Select the desired model.
3. Listen to the sound of the model name1. Open the model manager.

### Version

2.10.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

This OK
https://github.com/user-attachments/assets/1da1fdf8-132b-4702-91d4-c4395c47a7a8

This no good!

https://github.com/user-attachments/assets/3987ca1b-e7b5-439e-958e-aad05364650a

philmoz, 2024-07-17:
No issues on my TX16S.
Please zip up the IMAGES, MODELS, RADIO, SOUNDS, SCRIPTS, THEMES and WIDGETS folders from the SD card and upload here.

Plohish67, 2024-07-17:
> No issues on my TX16S. 
Thank you, I saved the data for you.
The rсgroups forum also confirms that there is a problem.
[image]
the file is made up of two parts, rename the file 222.zip to Sdcard.z01 and then open the archive
[Sdcard.zip](https://github.com/user-attachments/files/16267623/Sdcard.zip)
[222.zip](https://github.com/user-attachments/files/16267626/222.zip)

Plohish67, 2024-07-17:
Suddenly I solved the problem!
I thought that the SD card that came with the transmitter might be too slow, I ran a test and then I replaced the SD card with a faster one, the problem no longer occurs!
[image]

closed 2024-07-17 by 3djc