EdgeTX/edgetx #5300: protocols are changed

tonims1, 2024-07-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In 2.11 the protocols are changed in several models. I usually do it more in Frsky, going back to 10.2 doesn't happen

### Expected Behavior

that protocols are maintained

### Steps To Reproduce

select a model go to internal rf . the protocol has changed

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

pfeerick, 2024-07-18:
Changed from *what*? To *what*? Is it consistent? Have you updated the MPM firmware in the meantime, to a version of the firmware that doesn't support those protocols?

tonims1, 2024-07-18:
I only have Frsky and Flysky receivers. Those who are in Frsky change to Flysky but if you click to change it, the first option is the correct protocol. If you select it, change the receiver number to 0 as if you changed the protocol. If the protocol is Flysky 2, it becomes Flysky alone. Also, changing the model in the status module comes up randomly. no multi telemetry detected. until I turn it off and on again it doesn't detect it again. I have the latest version of Multi, 1.3.4.0 Air. if I go back to Edgetx 2.10. everything works fine again.
Being on the same model, I turn off the transmitter and turn it on again, it does a protocol scan and it is fine again, but if I leave the internal rf menu and come back in, it has changed again.

koaxheli, 2024-07-18:
Hello, I have the same problem with v2.11.0. The log is displayed with the wrong content when I select it.
If I go back to v2.10.2, everything is displayed with the correct names again.
v.2.11.0
[image]
Here is the 2.10.2
[image]

TomasJot, 2024-07-28:
I will ty to explain this bug further... After setting MULTI protocol and sub protocol in internal module configuration screen after  exit and entering again protocol and subprotocol displays incorrect names (eg. Hubsan instead of FrkskyV2). I traced this bug to  be introduced at commit b8dc505 - fix(color): wrong file chooser icon, portrait layout and other (https://github.com/EdgeTX/edgetx/pull/5201) (actually - more bugs was introduced, as inability to select protocol with scroll wheel, those was fixed at commit 0d9215a)

pfeerick, 2024-07-29:
Thanks for reports... the main part  - "protocols changed" is reproducible, and should be fixed now. The "no multi telemetry detected" is separate, and we'll have to see if it affects anyone else, otherwise it is most likely hardware or firmware specific.

PR #5338 (2024-07-29): fix(color): restore CHANGED event handler for Choice class
Fixes #5300
closed 2024-07-29 by pfeerick