EdgeTX/edgetx #5304: Companion X10S-E simulator - t5 and t6 sliders not working

sande005, 2024-07-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

T5 and T6 sliders do not change any values

Simulator only, work as expected in radio.

### Expected Behavior

That they would do so.

### Steps To Reproduce

Create model.
Assign T5 or T6 as source for Mixing.
Open simulator with channels/mixers also displayed
Note that sliders do not operate, but the up/down arrows for them do.  Cannot recenter with a double-click.

### Version

2.10.2

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

elecpower, 2024-07-17:
Cannot reproduce in Win 10. Arrows and sliders update Channel Monitor. Right double click re-centres.

pfeerick, 2024-07-18:
Cannot reproduce on Linux either... trim buttons, trim slider and right double click to reset all work. 
[issue5304.zip](https://github.com/user-attachments/files/16276154/issue5304.zip) (change file extension to `etx` after downloading to open in Companion)

@sande005 What OS are you using, as you omitted that detail.

sande005, 2024-07-18:
Win 11.
But....
I assumed, without testing further before posting.
[th-ele trim.zip](https://github.com/user-attachments/files/16277092/th-ele.trim.zip)
The attached is the prototyping I was doing to test out throttle to elevator mixing, that could be adjusted in flight.  T5 and T6 are not actually treated as sources, but are used in SF's to adjust GV's.  I assumed (the bad word) that I should be able to slide the sliders represented in the simulator, and double click to return to center.  After incorporating into a model on the TX, the real T5 and T6 don't slide - they are actually momentary switches, although specially set to act like potentiometers in some situations.  So Companion was actually behaving more like the real thing, despite the representation that they are sliders.
Sorry for the jumping to conclusions!

closed 2024-07-18 by sande005

pfeerick, 2024-07-18:
Oh, no, you didn't, did you? :rofl: Yeah, when using them as sources for GVs, or as the switch/trigger for SFs, the "position" of trim  becomes irrelevant... just the actual button press. But at least you figured it out ;)