EdgeTX/edgetx #5305: Unable to select the ASSAN protocol (and several others) in a 4in1 Radiomaster Tx internal Module

Framer91, 2024-07-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Similar to issue #3888 but workaround does not work. I have a TX16S i just upgraded from OpenTX to EdgeTX 2.10.2 and internal MPM to mm-stm-serial-aetr-lbt-v1.3.4.0.  My models using Assan receivers don't work any more. protocol 23 is "invalid protocol". If i try to redefine the protocol, only a few are offered (DSM, FrSky, Hott). i tried to set the module off then on, get MPM scanning progress bar complete but no change. If i choose external module MULTI all protocols are offered (i have no external module). I re flashed the MPM, no change.

### Expected Behavior

I would like all protocols listed in Multi.txt available, i mainly need Assan since i have a number of them.  Thank you.

### Steps To Reproduce

1. Open or create a model with  internal 4in1 Assan protocol. 2 Check settings in model setup page what protocols are available for internal module.

### Version

2.10.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Windows 10

### Anything else?

[image]
[image]
[image]

pfeerick, 2024-07-17:
Since you are using LBT MPM firmware type, it is possible that ASSAN is not
included in that firmware. Try using the FCC MPM firmware type instead.

closed 2024-07-18 by 3djc

Framer91, 2024-07-18:
Thank you. 
This defeats the purpose of the radio I bought precisely for bring able to use my Assan receivers. FCC IS illegal in France. So I guess I must back down the MPM code or even go back to Open TX. I do not know what is the risk of using an older MPM code. Edge TX said upgrade recommended, not mandatory.

3djc, 2024-07-18:
OpenTX would not make any difference. LBT MPM simply do NOT include Assan protocols at all

gagarinlg, 2024-07-18:
> Thank you. This defeats the purpose of the radio I bought precisely for bring able to use my Assan receivers. FCC IS illegal in France. So I guess I must back down the MPM code or even go back to [...]

When you use an older release it is as illegal as using the current FCC release.
The protocols were removed in the LBT/EU release because they do not use LBT, so they are illegal. Just nobody cared until some dealers got cold feed and RM asked Pascal to create a version of the firmware that only supports LBT compatbile protocols.

Framer91, 2024-07-18:
Thank you.
I went back to the previous MPM code and it does work. I thought an LBT receiver should/would not bind with an FCC transmitter and vice-versa . On FrSky receivers i have LBT firmware and my Assan receivers are supposed to be LBT (CE marked).

Framer91, 2024-07-18:
OK got it. I flashed with mm-stm-serial-aetr-air-v1.3.4.0 then select LBT (which is still there) for FrSky and it's all good.  It's appropriately called Air and not FCC. which i did not notice initially.
As for legal issues, pre 2015 bought radios are not forbidden, the receivers are pre 2015, so half and half...
Thanks a lot.

mha1, 2024-07-18:
> Thank you. I went back to the previous MPM code and it does work. I thought an LBT receiver should/would not bind with an FCC transmitter and vice-versa . On FrSky receivers i have LBT firmware and my Assan receivers are supposed to be LBT (CE marked).

Even if they were doing LBT (which I doubt) that doesn't mean MPM implemented LBT for this protocol. Example: The MLink protocol as implemented by Multipex is fully EU compliant meaning their radios as well as receivers are doing LBT. The MPM implementation of MLink works fine with Multiplex receivers but hasn't implement the LBT part. Now search for MLink in the non-FCC MPM firmware. You won't find it.

Framer91, 2024-07-18:
Thank you all.  
Hope this issue may help other Assan owners. 
Maybe a note in MPM.txt would help european users     for all protocols excluded from lbt code.