EdgeTX/edgetx #5307: Radiomaster boxer makes whining noise on 2.10.2

wlgrd, 2024-07-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Updating from 2.8.4 which my boxer came with to 2.10.2 causes it to emit a noise related to background light. 100% backlight causes the least amount of noise, and lower causes more until 0 where it is silent. Internal radio module turned off, did not help. Lipo at 8.3V. 

### Expected Behavior

No noise from background light

### Steps To Reproduce

* Install 2.10.2 on Radiomaster boxer.
* Radio starts emitting noise as long as the backlight is turned on

### Version

2.10.2

### Transmitter

RadioMaster Boxer

### Operating System (OS)

Linux

wlgrd, 2024-07-18:
Okay, I found a thread to check "Audio mute", but I don't understand what this does. I wouldn't want all my audio to be muted?

3djc, 2024-07-18:
Maybe you should try it ?

closed 2024-07-18 by 3djc

wlgrd, 2024-07-18:
Have had no use for audio yet, as I haven't even flown yet. But found 
```
Audio Mute - When enabled, puts the transmitter in mute mode until a sound needs to be played. This prevents interference noise from high-powered TX modules from coming out of the transmitter speakers
```
So I guess with this description I am fine.

pfeerick, 2024-07-18:
It mutes/shuts down the audio amp when the audio system is not actively playing sound, so you don't hear all the RF / electrical noise. Prior to 2.9, this was done all the time, but it was changed to being an option, which in hindsight should have been set to be by on default to keep prior behaviour. This was done as there is a delay between enabling audio and being able to then play sound - around 250ms - which is pretty crucial if you are using things like the variometer.