EdgeTX/edgetx #5310: Ability to quick select different models.

inventor7777, 2024-07-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am often switching between models on my TX16s and every single time I go to switch a model the model options menu (select model, duplicate model…) comes up first with no option just to skip that menu all together. (Other than pressing select model, obviously)

### Describe the solution you'd like

There should a at least a setting to skip that menu on a single tap/click and go straight into selecting the model. Since EdgeTX can understand long presses, I personally think only a long press should bring up the menu. 

(Since others may think differently, there could be a setting under Radio Settings like “Model Quick Select” or similar)

So you could just open the models page, tap the model, and you’re good to go.

### Describe alternatives you've considered

Don’t think there’s any other way to solve this.

### Additional context

Radio: TX16s Max MKii

I really only notice this issue on the TX16, as the UI is slightly more laggy than say, an MT12. On the MT12 I can pretty much double click on a model, on the TX16 it is slightly more laggy and I often click the wrong option 🤣

elekSENDER, 2024-07-18:
There is already a "Model quick select" chooseabel in Radio Settings behind "Manage Models".
As well the long press on the MDL button lead you directly in models menu.

so the quickest way is: (while Model quick select is "on"):
- long press button MDL
- doubletap the model

inventor7777, 2024-07-19:
@elekSENDER 
How did I not notice that 🤣 

Thanks very much!

closed 2024-07-19 by inventor7777