EdgeTX/edgetx #5312: Logic switch feature to allow division of total voltage telemetry (ie voltage divided by 4 for 4 cell lipo)

Dobin69, 2024-07-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Rather than using a widget like battAnalog, if logical switches had an option for division of an input ie total voltage telemetry divided by 4  in the case of a 4 cell lipo you wouldn't need the widget and could display the per cell voltage as a value field anywhere you like.

### Describe the solution you'd like

It should have the option of division by your choice of number of any chosen input.

### Describe alternatives you've considered

Using battAnalog - however the readout isn't clear in direct sunlight if you set the widget to the smaller sizes as the readout is over the top of colours of the battery display. Using the bigger size widget allows me to see it clearer but I like to use the space instead for lots of other telemetry readings in its place. 

Can't see any other alternatives.

### Additional context

The work you do is fantastic btw. Coming from Spektrum recently I'm blown away by the features edge tx has already. (Using tx16S)

pfeerick, 2024-07-19:
IMO... the better option would be
1) a calculated sensor (which you can use anywhere you can use telemetry sensors)... but for that to work *simply* it will need #2096, as then you will be able to use a GV to set your cell count, and then the calculated sensor would do the calculation based on your voltage sensor. 
2) a [Lua custom script](https://luadoc.edgetx.org/part_i_-_script_type_overview/mix) that does the calculation based on values you give it, and creates a telemetry sensor... which you can then use anywhere you can use telemetry sensors)
3) a [Lua widget](https://luadoc.edgetx.org/part_i_-_script_type_overview/widget_scripts) (although that would be colorlcd specific then)

Option 2 is pretty simple... and I even think I've seen this done before that way, but can't remember where :-/

3djc, 2024-07-19:
I run option 2 with a table that matches physical lipo testers so I have output in %, works great

mha1, 2024-07-19:
> I run option 2 with a table that matches physical lipo testers so I have output in %, works great

please share

Dobin69, 2024-07-19:
Thank you. Can you advise me how I can do option 2 please. I'm learning
quite a bit but that is unfamiliar to me so far.

3djc, 2024-07-20:
This is a very old script I use, should be updated. This is for FrSky FLVSS sensor, but it works with a total bat sensor. Name of your voltage sensor need to be changed in line 54

[pcalc.zip](https://github.com/user-attachments/files/16319097/pcalc.zip)

3djc, 2024-07-20:
[pcalc.zip](https://github.com/user-attachments/files/16319173/pcalc.zip)

This is a simplified version for non cells sensors.
Usage:
One time setup:
1) Edit the pcalc.lua file, enter your actual voltage sensor name and the desired name for the % sensor that will be created.
2) Put the file on sdcard in /SCRIPT/MIXES
3) Go to  CUSTOM SCRIPT screen and select the script (will need a firmware with custom script enabled )
4) Go to telemetry screen and scan for sensors, you will  get a new sensor with your desired name (BatP if you didn't change it)

You can then use your sensor like any other sensor (play audio value, use it for alarms,..)

Dobin69, 2024-07-20:
Thanks very much! I have done firmware updates and tinkered a little bit
here and there but how do I enable custom scripts pls? I will try this
today 😀

3djc, 2024-07-20:
If you use a tx16s, nothing, just enable the screen in radio setup/ enabled feature

Dobin69, 2024-07-20:
Brilliant thank you 😃

offer-shmuely, 2024-09-14:
I use a "mix" script to generate a sensor named "cell" that divide the battery by 3/4/5/6
see here:
https://github.com/offer-shmuely/edgetx-x10-scripts/blob/main/SCRIPTS/MIXES/cell.lua

it works, but it is very cumbersome. 
I do think that this a basic feature (especially with elrs, that we lost the FLVSS, and it always send a single voltage)
I love to see that implemented

this was my suggestion in the past:

[image]

[image]

[image]

Dobin69, 2024-09-15:
Thank you Offer 😃

bd4, 2024-09-15:
What about editing the telemetry sensor with a different ratio? Can do this by checking the battery voltage, plugging in and looking at the telemetry, then adjusting the ratio until it displays the correct single cell voltage. Can check again at the end to see if offset is also needed or if ratio is accurate enough.

This of course only works when you always use same number of cell battery for a model. I do have some planes I e.g. run on 4s and 3s, so having programmability would still be nice. Can detect the cell count at the beginning and set a GV accordingly.

inventor7777, 2024-09-16:
All great ideas. I’d love to have a programmable solution that can be easily adjusted and programmed to fit everyone’s needs without needing to calculate ratios.

Helipro, 2024-10-04:
> This is a simplified version for non cells sensors. [pcalc.zip](https://github.com/user-attachments/files/16319173/pcalc.zip)
> 
> Usage: One time setup:
> 
> 1. Edit the pcalc.lua file, enter your actual voltage sensor name and the desired name for the % sensor that will be created.
> 2. Put the file on sdcard in /SCRIPT/MIXES
> 3. Go to  CUSTOM SCRIPT screen and select the script (will need a firmware with custom script enabled (default for colorlcd and MT12)
> 4. Go to telemetry screen and scan for sensors, you will  get a new sensor with your desired name (BatP if you didn't change it)
> 
> You can then use your sensor like any other sensor (play audio value, use it for alarms,..)
> 
> Please note that "RxBt" in this script does only make sense with RM ELSR receivers that can measure Lipo voltage and return it in RxBt, in most cases, it will likely be more "VBAT" or something like that

Thanks so much man, been wanting this feature for long time.  I was able to edit this LUA, even though I cant write code, and return the cell count, and average voltage no matter what number of cells I put in.  I have a few planes that fly 3s/4s and 4s/5s.  So awesome to get my per cell voltage down to 0.00 with ELRS now.
Super excited thanks,
Craig

knputyi, 2026-01-30:
> > This is a simplified version for non cells sensors. [pcalc.zip](https://github.com/user-attachments/files/16319173/pcalc.zip)
> > Usage: One time setup:
> > 
> > 1. Edit the pcalc.lua file, enter your actual voltage sensor name and the desired name for the % sensor that will be created.
> > 2. Put the file on sdcard in /SCRIPT/MIXES
> > 3. Go to  CUSTOM SCRIPT screen and select the script (will need a firmware with custom script enabled (default for colorlcd and MT12)
> > 4. Go to telemetry screen and scan for sensors, you will  get a new sensor with your desired name (BatP if you didn't change it)
> > 
> > You can then use your sensor like any other sensor (play audio value, use it for alarms,..)
> > Please note that "RxBt" in this script does only make sense with RM ELSR receivers that can measure Lipo voltage and return it in RxBt, in most cases, it will likely be more "VBAT" or something like that
> 
> Thanks so much man, been wanting this feature for long time. I was able to edit this LUA, even though I cant write code, and add for it to return the cell count, and average voltage no matter what number of cells I put in. I have a few planes that fly 3s/4s and 4s/5s. So awesome to get my per cell voltage down to 0.00 with ELRS now. Super excited thanks, Craig [CelA.zip](https://github.com/user-attachments/files/18221035/CelA.zip)

Works like a charm!