EdgeTX/edgetx #5322: In Special Functions menu selection of SD Logs Function when enabled will intermittently freeze display

kwkenyon, 2024-07-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In Special Functions menu selection of SD Logs Function when enabled will freeze display often. The radio becomes unusable. You can disconnect battery to reset radio or hold the on button down for a few seconds.

### Expected Behavior

To not freeze display and allow radio to function. Radio must then be powered down removing battery or holding on button.

### Steps To Reproduce

When activating through Logical switch.

### Version

2.10.1

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

3djc, 2024-07-24:
Could you try another (genuine, high quality) SD Card ?

kwkenyon, 2024-07-24:
I will do that

kwkenyon, 2024-07-24:
Thank you, it seems the SD card works. I still had the original from RadioMaster, seems cheap. Thank you. I reckon you’ll close out the issue? Sorry for any inconvenience convenience.

closed 2024-07-24 by 3djc