EdgeTX/edgetx #5324: FLYSKY EL18 Buglist

fxk666, 2024-07-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

My friends and me had used 2.10.0 Few weeks.
sometimes we reaction a situation where the screen touch fails and has to be restarted.

### Expected Behavior

maybe someone met the same way as me.
who have the way to solve this problem. and give me your submission here！

### Steps To Reproduce

I have NO idea

### Version

2.10.2

### Transmitter

Flysky EL18

### OS Version

2.10.0 RC

### Anything else?

[image]

fxk666, 2024-07-26:
somebody help？

pfeerick, 2024-07-27:
@richardclli Your thoughts on this - didn't you mention there was some issue with the touch screen driver?

richardclli, 2024-09-02:
We are working to merge all Flysky radios into one target, all will be based on PL18 target.  Since touch driver is retouch and fine tuned in PL18 target, this may help the issues of NV14/EL18.  You may use this branch top build a firmware and help testing:

https://github.com/EdgeTX/edgetx/tree/fs_nv14_on_pl18

pfeerick, 2025-02-14:
@Bmfkangkang Can you try v2.11-rc2 when it is released (likely next week) and see if there has been any improvement?