EdgeTX/edgetx #5327: Slow response when changing number of points of a curve

fhpage, 2024-07-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Changing the number of points on a curve using either the wheel or touch commands is very slow to respond to changes. Can take several seconds (3-5). Using the roller wheel, have to turn one click and wait several seconds. Tested with 2.10.0, 2.10.2, 2.9.4.  Also tested with 2 TX16S and one X10S.  Not an issue with OpenTX on X12S. 

### Expected Behavior

The GUI should be able to follow the number of wheel clicks without stalling...

### Steps To Reproduce

Use scroll (roller) wheel to change number of points on a curve.

### Version

2.10.2

### Transmitter

FrSky X10 / X10S (ACCST), RadioMaster TX16S / TX16SMK2

TomasJot, 2024-07-26:
No problems on 2.11.0 - changing number of points takes < 0.5s

philmoz, 2024-07-27:
This is the performance issue with the text input fields in Lvgl.

@pfeerick - is it worth trying to back port the workaround from 2.11?

fhpage, 2024-07-27:
If 2.11 is a ways out to being released, a back port would be appreciated. This impacts everyone that uses curves. It is frustrating to use as is. Best alternative now is to use companion....

pfeerick, 2024-07-27:
Only you can answer that question Phil as you know how much code change is needed to make it work. I wonder if it is worth detaching that field so it doesn't "live update" on every field change, but rather on exit? Thus you only get one refresh update delay, not multiple as the values change?

philmoz, 2024-07-27:
It's also slow when changing from 'Standard' to 'Custom' because the point list gets rebuilt.

Back porting the 2.11 code would be a significant change (and affects all pages with text/number entry fields).
Since it's not technically a bug fix I wasn't sure about trying to retro fit into 2.10.

pfeerick, 2024-07-27:
That was what I thought might be the case... What about simply changing the field type - to a popup menu? thus it should only update once?

philmoz, 2024-07-27:
> That was what I thought might be the case... What about simply changing the field type - to a choice/popup menu? thus it should only update once?

That's probably the best option for 2.10.

I'm thinking this should also be applied to 2.11 as well for consistency.

pfeerick, 2024-07-28:
Agreed... Plus while 2.11 performance is better, it is still laggy...

philmoz, 2024-09-09:
Can this be closed?

PR #5333 (2024-07-28): fix(color): curve edit lag when changing point count, t15 graph size
Fix layout for T15.
Change point count value to a popup menu due to slow performance when rebuilding point list.

Fixes #5327
closed 2024-09-09 by pfeerick