EdgeTX/edgetx #5328: T16 w/o touch hard lockup after flashing master firmware

TomasJot, 2024-07-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Blank screen, hard lockup, have to disconnect battery. 

### Expected Behavior

Boot succesfull.

### Steps To Reproduce

Build from commit d0791fa - "chores: touch init in board (#5284)" on T16 w/o touchscreen.

### Version

Nightly (Please give date/commit below)

### Transmitter

Jumper T16

### Anything else?

Problem was introduced after moving touchPanelInit() from radio/src/opentx.cpp to radio/src/targets/horus/board.cpp
I guess on radios with touchscreen all woks fine, only on horus radio w/o touch it causes hard lockup...

pfeerick, 2024-07-27:
I can reproduce this with T16 - latest nightly 3d8854abdf7297491e64b99133daeeffdf67825c  as well as d0791fa0c22e71cb1b8f89c60e4ca96198eff847. Immediately prior commit b630982b20b6c1e14896f7b33a1cd0857ee5c392 boots fine.

3djc, 2024-07-27:
Don't have a T16, so tried to reproduce on TX16S and disconnecting the touch panel, unfortunately (well, kinda) it works fine

TomasJot, 2024-09-10:
Any luck with this issue? (I'm afraid to wear out battery connector of my T16...:)

TomasJot, 2024-10-09:
I still have a hope, so did some further digging... According to available schematics, T16 and TX16s have some hardware differences. On TX16S SCL and SDA signals to touchscreen connector have pullup resistors, on T16 those are  trim buttons. Other  2 signals on T16 not connected - so do not have pull ups too. 
Maybe it is possible to add cmake args -DHAVE_TOUCH=YES/NO for T16?

raphaelcoeffic, 2024-12-01:
> Other 2 signals on T16 not connected - so do not have pull ups too.

Might indeed be related to pull-ups. We should probably check the internal pull-ups are in case no external ones are populated.

TomasJot, 2024-12-05:
Is there something I can do to help fix this issue?

pfeerick, 2025-01-24:
Yes. Latest nightly still results in T16 not booting - only the 6POS array flashing, so reverted it back to 2.10. One additional point - I DFU flashed so it had the matching bootloader, and I *was* able to enter the bootloader, but FWIW entering via the back door (Exit) did not work either.

pfeerick, 2025-02-05:
Just tried again with latest nightly, radio still locks up on boot. A revert of https://github.com/EdgeTX/edgetx/commit/d0791fa0c22e71cb1b8f89c60e4ca96198eff847 still works ... wait... `touchPanelInit()` is being called twice now. 🤦‍♂ Removing that seems to fix it.

PR #5868 (2025-02-09): fix: remove redundant touchPanelInit
Appears the (unintentional, as it is already done in board.cpp a few lines further down?) double init introduced in #5284 locks up T16 hard on boot if no TP present. Tested TX16S w/ TP, still working fine. 

Fixes #5328
closed 2025-02-09 by pfeerick