EdgeTX/edgetx #5330: T15 - Lua getValue() rerturns wrong value

JimB40, 2024-07-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

for sources thr, ele, ail, rud Lua getValue('thr') rerturns always 0 

### Expected Behavior

returns proper value <-1024;1024>

haven't tested all arguments for getValue()

### Steps To Reproduce

```
local run = function(evt, ts)

  local thr = getValue('thr')

  lcd.clear()
  lcd.drawText(10, 10, 'THR: '..tostring(thr))

  return (evt == EVT_EXIT_BREAK) and 1 or 0
end

return { run=run}
```

### Version

2.10.2

### Transmitter

Other (Please specify below)

### Anything else?

T15

3djc, 2024-07-27:
Reproduced

closed 2024-07-27 by pfeerick

JimB40, 2024-07-27:
@3djc same but with trims soursces trim-thr, trim-ele ...
If code was changed, I think you should make some tests changing lua source for at least for one group member ie. inputXX, gvarX etc.

pfeerick, 2024-07-29:
> @3djc same bug with trims sources trim-thr, trim-ele

This was fixed in #5331 also. 

```lua
local valuesToTest = {
  'ail',
  'ele',
  'thr',
  'rud',
  'trim-ail',
  'trim-ele',
  'trim-rud',
  'trim-thr'
}

local run = function(evt, ts)
  lcd.clear()
  
  for i, valueName in ipairs(valuesToTest) do
    local value = getValue(valueName)
    lcd.drawText(10, 10 + (i - 1) * 15, valueName..' : '..tostring(value))
  end

  return (evt == EVT_EXIT_BREAK) and 1 or 0
end

return { run=run }
```

[image]

PR #5331 (2024-07-27): fix: lua axis name for T15, T14, T12Max
Fixes #5330

Summary of changes: add missing lua names for axis