EdgeTX/edgetx #5332: Taranis X-Lite trim switch does not match stick mode

chrisu365, 2024-07-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Have a look to https://github.com/opentx/opentx/issues/8260
Setting mode 2 stick configuration the trim switch works for left stick without shift and for right stick with shift key. 

### Expected Behavior

Trim should  follow the stick mode setting so that the "unshifted" trim keys always trim aileron and elevator as these are the most critical trims

### Steps To Reproduce

1. set transmitter stick mode to 'mode 2' (aileron on right stick)
2. try left/right trim --> it takes effect on left stick
3. try up/down trim --> it takes effect on left stick
4. set transmitter stick mode to 'mode 4' (aileron on left stick)
5. try left/right trim --> it takes effect on left stick
6.try up/down trim --> it takes effect on left stick

### Version

2.10.2

### Transmitter

FrSky X-Lite / S / Pro

### Operating System (OS)

Windows

### OS Version

Windows

chrisu365, 2024-09-25:
As i am using stickmode 2 only, i modified KEY_SHIFT using the following code:
https://github.com/EdgeTX/edgetx/tree/main/radio/src/boards/generic_stm32/inputs.cpp
org:
#if defined(PCBXLITE)
if (_read_keys() & (1 << KEY_SHIFT))

new:
#if defined(PCBXLITE)
  if ((_read_keys() & (1 << KEY_SHIFT))==0)

markran, 2024-10-01:
I just posted on RCG asking about this and was referred here. Happy to see it was already reported. Hopefully, this regression will be addressed shortly!

pfeerick, 2024-10-04:
@markran As an X-Lite owner (presumably), would you be able try PR #5580 PR, which hopefully fixes this on hardware 🤞 

> To access the automatic builds for a Pull Request (PR) you are interesting in trying out, first open up page for the PR in question. Under the title for the pull request, you should see a series of tabs (Comments, Commits, Checks, Files Changed). Click on the Checks tab. 
> 
> If you are after the radio firmware, click on the link on the left side mentioning Firmware. 
> 
> Then, under the 'Artifacts' heading near the bottom of the page will be a link to the zip file containing either the firmware or companion package.

chrisu365, 2024-10-07:
I installed the artefact from: https://github.com/EdgeTX/edgetx/actions/runs/11134881058/job/30944247037?pr=5580
It works on my real X-Lite as expected
Thanks a lot !

PR #5580 (2024-10-08): fix(bw): unshifted trims on XLite radios should always adjust elevator and aileron
Fixes #5332 

Restore OpenTX functionality.

Note: not tested on actual radios, only in simulator.

Could be applied to 2.10.5.
closed 2024-10-08 by pfeerick