EdgeTX/edgetx #5344: Enable/Disable Buttons on the radio with a switch

AlphaCharlyX, 2024-07-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Sometimes it could be usefull to be able to disable the buttons on the radio. Especially if you have big uand like me. On me Radiomaster TX12 I often push the Model- or Sys-buttons accidentally.

### Describe the solution you'd like

There is a option to lock the Touchscreen with a switch. The same option could be integrated in EdgTX for locking the buttons of the radio during flight or if you give the radio to a student.

pagrey, 2024-08-13:
Disabling hardware controls like the SYS or MDL buttons could render a radio unusable until it is connected to a computer or had a new SD card inserted. People like to play around with settings and having one that locks you out might not be a good idea. I like the "demo mode" concept but it might cause some headaches.

pfeerick, 2024-08-14:
That is my concern here also... since for _most_ radios, touch is one of two input methods. For this to work, there would perhaps need to be a bypass... i.e. perhaps if you hold the enter key for 5 seconds when the key lock SF is active, the keys are bypassed. Something similar is probably needed for touch also... when press on the top left (i.e. where the ETX logo is *usually* displayed) for more than five seconds, touch screen lock is disabled (of particular importance for those radios only with touch screen interfaces).

JOHNJETY, 2025-05-03:
I broke my Very expensive Vortex3 DLG glider yesterday due to this issue. In between landing and launching i somehow inadvertently inverted the elevator, I did have the Tx around my neck, but took it off to hold it in my hand for the launch and in doing so I must have pressed some buttons/touchscreen.
 The Tx was on the ‘default screen’ as normal but ended up displaying mixer graphs.

pagrey, 2025-05-03:
Sorry to hear about your glider. You can disable touch with a special function but that doesn't do much for the keys.

If you create a full screen widget selecting App mode instead of full screen the widget now accepts key input. Just select a simple widget that displays the model name. When you select the widget page and long press enter all the inputs will be disabled unless you long press the exit key. It seems to me that's a decent screen lock. You could even write a simple lua script widget that says lock if you want or beeps if you press any key so you know you are touching buttons when the screen is locked.

bigtwin, 2026-05-18:
@pagrey wrote "_As mentioned above, using logic or switches to disable navigation could lead to a lockout, that's not great._"

I dont think this apply here. It is not about entire _navigation_. If you just disable individual buttons (like e.g. the sys-button), there is no risk of a lock-out. Of cause, unless a software failure results into a total lock-out. And the button-disable can be release by several events, like a "model change" or pressing "model settings" ...etc.
It might be an idea to even allow disabling individual buttons by _Special Functions_ so everbody may define his own preferences.