EdgeTX/edgetx #5347: Issues with latest Nightly

inventor7777, 2024-07-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I recently updated my TX16s Max to the 2.11.0 beta (the most recent one as of now) and I immediately noticed two issues.

#1 - Value2 completely stopped working completely. I had some set up on my main screen and they completely disappeared (I even tried updating the widget pack with Buddy, but no go). The folder is still there “Value2” and I even tried the version from my last backup.

#2 - The startup delay is absolutely terrible. I used to just have to hold the button until the orange lights went from left to right to left to right, then the radio would power on. Now I have to wait for the lights to go left right left right left.

### Expected Behavior

I expected my widgets to stay exactly how I left them and for the startup time to remain the same.

### Steps To Reproduce

Update a TX16 which is using Value2 widgets and let the drama begin.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

macOS

### Anything else?

I also noticed that the USB and Power Off screens now seem to use a different source for their coloring (the Carbon Lime theme makes the background bright lime, while my theme (Carbon_Red) reverted to an all black background.

pfeerick, 2024-07-29:
I believe (but am not certain) the USB screen may be fixed colours. Power
off follows the theme colours, so is dependent on what colours you used
used where ;)

philmoz, 2024-07-29:
The Value2 widget is Lua and will need to be updated by the author - it checks the EdgeTx version; but has no error checking for new versions (since 2.11 is not in the array it tries to access a nil value):
```Lua
local function getSensorId(key)
    local OS_SWITCH_ID = {
        ["2.7"]  = {RSSI=253, CELS=0  , TxBt=243, RxBt=256},
        ["2.8"]  = {RSSI=261, CELS=300, TxBt=251, RxBt=0  },
        ["2.9"]  = {RSSI=294, CELS=121, TxBt=251, RxBt=291},
        ["2.10"] = {RSSI=0  , CELS=0  , TxBt=0  , RxBt=0  },
    }
    local ver, radio, maj, minor, rev, osname = getVersion()
    local os1 = string.format("%d.%d", maj, minor)
    return OS_SWITCH_ID[os1][key]
end
```

philmoz, 2024-07-29:
Both of the USB connected and Shutdown screens use theme colors (PRIMARY1 and PRIMARY2).

pfeerick, 2024-07-29:
Offer's Flights widget has the same issue :(

inventor7777, 2024-07-29:
@philmoz and @pfeerick - thanks for the help! Do you guys know why the startup time has been increased on the TX16s?

@offer-shmuely - when you update these widgets, maybe you could add a failsafe into the widget next time? Maybe just some extra code where if the OS version of the radio is not on the list, just use the values of the last version?

I was able to get the widget working again by simply copy-pasting “["2.10"] = {RSSI=0  , CELS=0  , TxBt=0  , RxBt=0  },” and changing it to “["2.11"] = {RSSI=0  , CELS=0  , TxBt=0  , RxBt=0  },”.

pfeerick, 2024-07-29:
Not sure on that one... it may be another side effect of adding the H7 support. Although switching between 2.10.2 and nightly just now... I don't see any real difference in startup speed. 

Have you still got the splash screen time in radio setup set to 4s (4 seconds?) Try bringing it down to 1s... i.e. remove the artificial splash screen delay. ;)

inventor7777, 2024-07-30:
I had it set at 1 second before. I definitely can tell a difference though. I am especially confused since I thought the power on delay (the orange light cycling) was all hardware based? Bizarre…

pfeerick, 2024-07-30:
Indeed, the crows feet 6POS led startup animation is independent and nothing to do with radio firmware... so you can just us it as a timing reference. Only thing I can suggest is try a different SD card... perhaps the one in the radio is slow or there is a file system fault... 

Then again, what version of ETX did you upgrade *from*? As it could be a slowdown that was introduced a little while back and probably isn't going away any time soon :zany_face: 2.10.2 gives me the splash screen as it is about halfway through the third pass of the crows feet animation, whereas 2.11 is nearer the end of the third pass... and that sort of variation isn't anything special. (btw, I'm counting a pass as the complete cycle from 1 to 6 and back to 1 again)

philmoz, 2024-08-01:
Startup time to when the main view appears should be faster in 2.11.

closed 2024-08-06 by inventor7777

Mmmonsterkill472, 2024-10-10:
++