EdgeTX/edgetx #5349: FrSky Xlite missing internal XJT module option

dan-sergison, 2024-07-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

2.10.2 centurion.  Frsky Xlite.   Hardware menu selection Internal RF only has Multi or CRSF, no XJT choice.  Can select multi then in a model set the type to Frsky D8 but still can not bind.  my Rx's are all on ACCST

### Expected Behavior

Josh Bardwell says you select XJT in the hardware menu, but option does not exist, only CRSF or Multi.   Trying multi here in radio hardware lets me select frsky d8 in the model setup but it still does not bind.

### Steps To Reproduce

menu to radio hardware.
internal RF
Type - set to (missing XJT?) only CSRF or Multi 

### Version

2.10.2

### Transmitter

FrSky X9 Lite / Lite S

pfeerick, 2024-07-29:
This was fixed in #5262, which will be part of 2.10.3, likely to release this week. In the meantime, you can either follow the instructions here https://github.com/EdgeTX/edgetx/issues/5261#issuecomment-2212418812, or flash the latest 2.10 branch firmware. The former is probably the simplest, as you should only need to do it once.

closed 2024-07-29 by pfeerick

dan-sergison, 2024-07-30:
you are super awesome.  I've been in firmware hell and sure I had done something wrong and feeling very dumb.  and you wonderful dev's already fixed it.  I just flashed the nightly 2.11...  and it works great. Thank you so much.

pfeerick, 2024-07-30:
Glad to hear it... enjoy! :)