EdgeTX/edgetx #5350: DSM GPS telemetry not working when using the SKY_ID sensor

frankiearzu, 2024-07-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The TX is not getting the GPS data on the traditional GPS telemetry data frames..  It used to work, but now is no longer working (2+ people report it).

Suspect that they are probably migrated the GPS DATA to be part of the SKY_ID telemetry frames, and turned off the old ones.

 

### Expected Behavior

Correctly report GPS data.

### Steps To Reproduce

Updated firmware on RX  (2.48 or new 3.x).
Using SKY_ID sensor.

### Version

2.10.2

### Transmitter

RadioMaster TX16S / TX16SMK2

frankiearzu, 2024-07-30:
@qsiguy, @TEH-ArcGIS
Based on your captures, was able to see that they are embeding the GPS telemetry packets inside the RemoreID telemetry packages.. exactly the same format, but they are hacking a bit the protocol.

Already did the change to the firmware.  What radio and EdgeTX version are you using?? will create a firmware for you guys to test..  I test it with my own testing framework that i added before by copying the captured data.

frankiearzu, 2024-07-30:
Example capture:  
27 is the Remote ID sensor, 17 is the GPS Stats, and 16 is the GPS Loc.
They are using the 2nd byte that is the "Instance" of the sensor, and piggyback to add the GPS data.
--------- Capturing I2C_ID 27
27 17 00 00 10 17 08 01 07 00 00 00 00 00 00 00 20 88     -- GPS Stat
27 27 01 60 31 38 34 39 45 30 43 43 42 35 34 43 20 88     -- True RemoteID info
27 16 36 48 28 96 09 33 72 16 35 11 00 00 28 3D 20 88    -- GPS Loc
27 17 00 00 10 18 08 01 07 00 00 00 00 00 00 00 20 88     -- GPS Stat
27 16 36 48 28 96 09 33 72 16 35 11 00 00 28 2D 20 88
27 17 00 00 10 18 08 01 07 00 00 00 00 00 00 00 20 88
27 17 00 00 10 18 08 01 07 00 00 00 00 00 00 00 20 88
27 27 01 80 34 33 43 46 31 36 30 00 00 00 00 00 20 88

qsiguy, 2024-07-30:
@frankiearzu I am currently running 2.10.2 firmware.

TEH-ArcGIS, 2024-07-30:
Good morning, Frankie,

My operating system is 2.20.2 "Centurion"

Looks like qsiguy and I have followed similar testing steps.

Thank you for taking another look at this....I can't put my finger on the issue???

***@***.******@***.***>
515.664.7674
 Population is growing @ 1.1%/year
Food demand is growing @ 1.75%/year
Energy demand is growing @ 2.0%/year
Water demand is growing @ 2.7%/year

frankiearzu, 2024-07-30:
Can you please test this TX16S firmware.  Is compiled using the current 2.10.x branch + this fix.  This will become 2.10.3 (so 2.10.2 + a few other fixes).  Please report if now the SkyID is looking good.

https://github.com/frankiearzu/DSMTools/blob/main/EdgeTx_Firmware/TX16S_2.10.2_SKYID.bin

qsiguy, 2024-07-30:
Excellent, I've got data streaming!  I'll send you a video of my values directly but looks good.  I see you eliminated the 27xx and 05xx values?  List is much shorter but those were pretty much useless values I think, at least I never knew what to use them for.

closed 2024-07-31 by pfeerick

TEH-ArcGIS, 2024-08-02:
Frankie, SKY_ID.bin patch works!!!  I have GPS LAT/Long telemetry!

Thank you!!!

https://github.com/frankiearzu/DSMTools/blob/main/EdgeTx_Firmware/TX16S_2.10.2_SKYID.bin

frankiearzu, 2024-08-04:
Just FYI, this change was included in 2.10.3.. so no need for custom/self-build firmware any more.. will delete it from my site.

TEH-ArcGIS, 2024-08-04:
Good to know!  I was wondering.

Frankie thanks for your help and time!!!!

PR #5355 (2024-07-31): feat(dsm): process new format for Spektrum Sky_ID GPS frames
Fixes #5350
Tested with the 2.10 branch as well as main.

Summary of changes:
1. Looks like in the latest RX firmware, Spektrum changed the way the SkyID GPS data is sent on the telemetry. Originally, the SkyID GPS data was sent using the old/standard GPS Frames, but now is not doing it.  The GPS data now is embeded inside the RemoteID telemetry frames.. exact same format, so the existing parsing can be re-used.

3. A few people on Black&White radios has complained that the 40 telemetry sensors is not enough for for some of the Spekt [... cut]