EdgeTX/edgetx #5368: monocrome display (212x64)  doesn´t display in custom/telemetry screen 4th line in "type=numbers" mode

AirWalt, 2024-08-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On monocrome display (212x64) in EdgeTX as well as in Companion both running v2.10.2 AND also under brand new v2.10.3 shows no 4th line in custom/telemetry screens in type=numbers mode. 4th line can be configured in Tx and/or in Companion but will not be displayed on TX nor in Campanion simulator.
Ocurred/tested on FrSky Taranis X9D+ 2019 SE .

### Expected Behavior

Display in telemetry screen 4th line in type=number mode. Or if not applicable disable the configuration of 4th line in type=number mode in EdgeTx/Companion.

### Steps To Reproduce

  Configure 4th teelemetry line in type-numbers mode.

### Version

2.10.2

### Transmitter

FrSky X9D+2019

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro

pfeerick, 2024-08-03:
No, the bottom line changes from saying "no data" (if you don't have telemetry), to being the 4th line. 
i.e. no telemetry:
[image]
with telemetry:
[image]

closed 2024-08-03 by pfeerick

AirWalt, 2024-08-03:
Ah - tnx for the explanation. I expected a similar line hight as for line 1-3  ;-)