EdgeTX/edgetx #5378: No function external MPM as trainer module on X10s/X9 lite

elekSENDER, 2024-08-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I tried the new feature "external MPM as trainer module" on 2 FrSky radios (X10S with ISRM upgrade module, X9 lite). Binding the student radio, in my case a Spektrum DX9 (DSM RX configured in the external MPM), is only possible by switching off the internal RF.
After successfully binding the student to the trainer i can see the expected functionality in the channel monitor on the trainer.
By switching on the internal ISRM module the trainer connection is immediatly lost and no student inputs are seen in the monitor, switching off the internal module you can hear immediatly "trainer connection recovered" and the student inputs are back.
Same behavior on both radios.

Maybe I´m just unlucky and only my 2 radios are affected of this behavior?
Or is there anything more to adjust in comparision to a TX16s?

FYI: Working with 2 RX simultaneously works perfect, i tried this intern with an archer plus and extern with DSMX, Flysky and Traxxas RX, no issues on both radios. Great job!

### Expected Behavior

Student inputs working with internal RF switched on

### Steps To Reproduce

Make a trainer/student configuration with a trainer FrSky(?) radio 
Bind a student radio
Check student functionalty in monitor
Switch on/off the internal RF

### Version

2.10.3

### Transmitter

FrSky X9 Lite / Lite S, FrSky X10 / X10S (ACCST)

pfeerick, 2024-08-08:
ISRM (ACCESS) transmitters behave differently to XJT (and all others), so this if this is the problem I think it is, this can't be resolved by us as it is an issue with the radio hardware itself. Normally you might have been able to disable the MPM telemetry, in order to use both RF modules simultaneously, but if you are trying to use any of `_RX` modes, you can't, as you *need* that telemetry for that mode to work. You can use ISRM and MPM simultaneously for all the other "normal" transmit modes, and you can also toggle the "Disable Telemetry" for the external MPM in that case, as it will not work anyway with the ISRM active, and will prevent "telemetry lost" nags about the MPM. 

The "new feature "external MPM as trainer module" is more for transmitters that have an internal MPM, as there appears there was a bug in how the telemetry was being handled... as being able to use dual MPM is not actually a new feature... has been around some time ;)

elekSENDER, 2024-08-09:
ah, ok, understood.

yes, i know the trainer mode with external MPM, I use that for trainer-student since it was possible on a TX16S, realy like it.
And now i thought its also possible with ACCESS radios.

Thx for your detailed explanation.

closed 2024-08-21 by pfeerick