EdgeTX/edgetx #5393: Telemetry

HTTra, 2024-08-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

RadioMaster Pocket Remote Control does not have a telemetry sensor that displays GPS coordinates and number of satellites in version 
EdgeTX v2.10.3

### Expected Behavior

Mong sớm khắc phục sự cố

### Steps To Reproduce

Telemetry

### Version

2.10.3

### Transmitter

RadioMaster Pocket

inventor7777, 2024-08-06:
Have you gone to MDL-Teelemetry-Discover new? @HTTra

frankiearzu, 2024-08-07:
What GPS sensor and Receiver are you using?

pfeerick, 2024-08-07:
Again,

> What GPS sensor and Receiver are you using?

It seems like there is no data coming in for the GPS (no star strobbing/flashing to indicate updates)? Without knowing how/where this data is coming from?, when it worked last?, what has changed in the meantime?, etc, it is hard to diagnose / help you. e.g. If this is coming from a GPS connected to a flight controller, perhaps the flight controller is mis-configured?. GPS has become disconnected? Is there a light on the GPS to indicate it is getting power / has a position lock?

HTTra, 2024-08-07:
> điều

Tồi dùng phiên bản edge tx v2.10.0 thì lại có senso gps

HTTra, 2024-08-07:
> Lại,
> 
> > Bạn đang sử dụng cảm biến và bộ thu GPS nào?
> 
> Có vẻ như không có dữ liệu nào được gửi đến GPS (không có sao nhấp nháy/nhấp nháy để báo hiệu cập nhật)? Nếu không biết dữ liệu này đến từ đâu/bằng cách nào?, lần cuối cùng nó hoạt động là khi nào?, trong thời gian đó đã có thay đổi gì?, v.v., thì rất khó để chẩn đoán/giúp bạn. Ví dụ: Nếu dữ liệu này đến từ GPS được kết nối với bộ điều khiển chuyến bay, có lẽ bộ điều khiển chuyến bay đã được cấu hình sai?. GPS đã bị ngắt kết nối? Có đèn nào trên GPS báo hiệu rằng nó đang được cấp nguồn/có khóa vị trí không?

Khi tôi thay đổi phiên bản EdgeTX v2.10.0

Uploading .lv_0_20240807090558.mp4…

pfeerick, 2024-08-07:
1. What GPS sensor and Receiver are you using?

2. I understand. 2.10.0 works. Does 2.10.2? There were some Spektrum GPS changes in 2.10.3.

HTTra, 2024-08-07:
> 1. Bạn đang sử dụng cảm biến và bộ thu GPS nào?
> 2. Tôi hiểu. 2.10.0 hoạt động. Còn 2.10.2 thì sao? Có một số thay đổi của Spektrum GPS trong 2.10.3.

Tôi sử dụng gps CCRC M10 và thay các gps khác vẫn không có telemetry. Phiên bản edge tx 2.10.2 thì vẫn không hoạt động.