EdgeTX/edgetx #5394: Multi Position Mode

inventor7777, 2024-08-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Hi all!

I have an MT12 and I want the SB switch to be able to switch between several different options (like STKY mode, but it cycles between more than two options.) I thought there was already a way to do this, but I can’t figure it out.

### Describe the solution you'd like

There should be some option (each model needs to be able to configure this separately) for a “cycle” or similar option under Model settings where you can turn a standard switch into a multi position switch. E.g choose a switch, choose the number of positions it has, then it would show up as a multi position switch for logical switches and anywhere else.

### Describe alternatives you've considered

Don’t think there is any other way to get the desired result.

### Additional context

I thought for sure that there was an option for this already, but I could not figure it out all all. I also need this to be applied only per model, as I have some cars that need that switch to stay as a 2pos.

If this is implemented, (similar to the STKY) there could be a persistence option so the radio remembers the last position of this software switch.

pfeerick, 2024-08-06:
You should be able to use a global variable (GV) and use the adjust SF to
increment it on each press of SB. Then use LS to detect when GV is a given
value, for use in the mixer as the on condition. And perhaps another LS and
SF to reset back to first position once done the full walk. Otherwise you
could probably do more complex logic to have it go backwards as well.

inventor7777, 2024-08-07:
@pfeerick how do I make it advance? It only seems to set the GV to the set value, and subsequent pushes do nothing. Any ideas?

inventor7777, 2024-08-07:
Never mind….figured it out! I didn’t even know EdgeTX had that option 🤣

pfeerick, 2024-08-07:
:thinking: Checkout "Adjust Global Variable" => https://manual.edgetx.org/bw-radios/model-select/special-functions

But yeah, that plus setting boundaries for your GV - i.e. min and max values, can be pretty powerful... can do a lot with GVs, LSs and SFs... without resorting to lua scripts for more complex stuff.

inventor7777, 2024-08-07:
Nice! I missed that part when I searched the manual for that yesterday :)

Thank you very much!

closed 2024-08-07 by inventor7777