EdgeTX/edgetx #5400: Menu bar has disappeared

koaxheli, 2024-08-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

The new version of the TX16S transmitter is v2.11.0-main (ca459bb9)
The problem is when I open the telemetry display data in edit mode with SYS and close it again, go back to the main menu the menu bar is dead.

### Expected Behavior

Since the change and the addition of the full screen app there is a lot of trouble.

### Steps To Reproduce

Switch to telemetry and go back

### Version

2.10.3

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

Before
[image]
After
[image]
[image]

Kevltan, 2024-08-07:
Had the same problem as you. Was just compiling the issue as you put yours in😂😂😂

philmoz, 2024-08-07:
I cannot reproduce this. Please provide more details on how to reproduce it, or upload your RADIO, MODELS and SCRIPTS folders in a ZIP file.

![screenshot_tx16s_24-08-07_12-42-50](https://private-user-images.githubusercontent.com/92114465/355790650-79b07969-182a-4c41-a866-ead9fd5741bb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwNjc3OTYsIm5iZiI6MTcyMzA2NzQ5NiwicGF0aCI6Ii85MjExNDQ2NS8zNTU3OTA2NTAtNzliMDc5NjktMTgyYS00YzQxLWE4NjYtZWFkOWZkNTc0MWJiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA3VDIxNTEzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMyYzFlMDZmZjIyZjcwZTM3MjY0NDJhMGQ0OTRkOWYzOTgxMTA5MTAxOGQ4NWVkODZlZDBmNDQ5NTkwOTEzOWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.UOv7HP5GRpwy3XV0Q9-Qg2YCXriNDkg6Hiu6dqmU9Rc)

There is something very wrong here
- the date & time are clipped
- the selected tab icon in the menu does not match the page content
- the Layout does not show a layout preview image

pfeerick, 2024-08-07:
> * the Layout does not show a layout preview image

Not quite... it does - it's showing the "full screen" layout, but the button isn't wide enough. 

> telemetry display data in edit mode with SYS and close it again

Not quite sure what you are getting at here... telemetry in model settings, not SYS? Exact sequence of steps is probably needed, unless the issue is evidence from the zip of model/radio settings. Will also need the WIDGETS folder in addition the others, so we've got the right version of the widgets in use (i.e. showitall).

philmoz, 2024-08-07:
> > * the Layout does not show a layout preview image
> 
> Not quite... it does - it's showing the "full screen" layout, but the button isn't wide (and perhaps tall) enough.

But the button should auto size and look like this (something is very screwed up to create the image posted above):
[image]

pfeerick, 2024-08-08:
True, but doesn't really change the fact the layout preview **is** being shown... :stuck_out_tongue: There is no denying some weird stuff is going in hereAt least there are two separate reproductions of this behaviour, so that should hopefully make it easier to pin down the cause(s). 

Same as I had the TX16S do a full screen dialog a little with the background missing (I could see the 6POS buttons from the main view) a little while ago... but do you think I can get it do do it again to work out the reproduction steps? :rofl: I *thought* it was some was combination of model select and screen settings, but it could also have been model setup / turning on MPM (i.e. the delayed "failsafe not set" warning)...

koaxheli, 2024-08-08:
Sorry, I meant MDL button.
Press MDL button then go to telemetry open any file with EDIT-close go back and the menu bar is gone.
When the TELE button is pressed the view is as in screenshot_tx16s_24-08-07_12-42-50 and unchangeable.

Kevltan, 2024-08-08:
Yes agree with above after opening a telemetry field and then exiting to model page I lose the model name input window on the right and end up with a small vertical window on the left, where you can just make out one character and the top bar freezes with date and time corrupted. This is when using a custom fullscreen widget.
When using a "normal" model (no custom full screen widget) with standard accessable widgets, I lose the top bar completely when exiting from telemetry page.

pfeerick, 2024-08-08:
Same here also, New blank model, #5386 build on TX16S. Going Model Settings -> Telemetry -> Add new -> RTN -> RTN results in loss of top bar. Going into screen settings either via carousel menu or TELE key after that results in the aforementioned screenshot with clipped time, mangled layout item. Something in the sensor edit screen edit screen sets this off... all is good until then.

koaxheli, 2024-08-08:
Here are the MODELS, RADIO and the Edge bin version v2.11.0-main ( ca459bb )
[ETX_2.11.0.zip](https://github.com/user-attachments/files/16546700/ETX_2.11.0.zip)

koaxheli, 2024-08-08:
Yes, everything is OK at the transmitter.
Companion still faulty.

Kevltan, 2024-08-08:
> results in the shown screenshot with clipped time, mangled layout item.

Was this fault repaired as ticket closed??

pfeerick, 2024-08-08:
No, fix hasn't been merged yet as I was sleeping while it was fixed 😂 The damage one little missed line can cause!

PR #5408 (2024-08-08): fix(color): view refresh issues after opening telemetry sensor edit page
Fixes #5400

Enable LVGL refresh after building sensor edit page.
closed 2024-08-08 by pfeerick