EdgeTX/edgetx #5403: GV does not use correct value when used as a source

sande005, 2024-08-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

When an SF is used to record a GV from another source, and then that GV is used as an Input source, the result is 1/10th of the original.
Ex: if Ele is -68, and recorded to GV1, then GV1 is used as a mixer replace source, it results in a value of -7 (6.8 rounded) being used, not -68.  Changing units/prec in GV settings will change the display value of the GV, but not the end result.

This has not been tested in radio, to see if behavior is different.

### Expected Behavior

Maintain the actual value.

Simple version of issue attached.  Change 'txt" to "etx"
[GV Replace Input.txt](https://github.com/user-attachments/files/16532465/GV.Replace.Input.txt)

### Steps To Reproduce

Use a SF to adjust a GV to a Source value.
Define the GV as an Input
Set up a mix replace line to use that GV Input as a source.

### Version

2.10.3

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

### OS Version

Win 11

philmoz, 2024-08-07:
The value saved to the GV by the SF is not the raw source value; but the scaled % value (-100 to 100).

Changing this would likely break existing setups.

It might be possible to add another mode that copies the raw source value.

sande005, 2024-08-08:
Um...OK....that explanation flew well over my head.  But I'll do my own research to better define the terms to me and how they relate...but I can appreciate that there is more going on behind the curtain than it appears.

pfeerick, 2024-08-08:
When you do the Adjust GV from Source, it captures the value scaled from -100 to 100, as displayed in the mixer. 
But when you use the GV as a source, it is expected to be in the -1024 to 1024 range. 

Now, you *can* increase the weight to 500 on the mixer, but that will only get you to -49%/49% (1250us/1750us). So something more is needed. 

Perhaps another Adjust GV from Source mode is needed, one being Scaled (the current), the other Raw?

sande005, 2024-08-08:
Thank you pfeerick - I just found that explanation in a lua doc example for getSource(), also.  Makes sense.
In my case, this is just an exercise to see if I could dual purpose the ail/ele stick to also be a camera pan control on a joystick.  So am scheming to have it do a "last position hold" when that function is activated.  There may be other ways to skin that cat, without changing the code....
It looks like using GV's as Sources is an Edge improvement (not in OTX 2.3.15), perhaps a selectable option to upscale the GV*10.24 first, when used as a Source (where the existing code would then downscale it again) - kinda kludgy, but..??

pfeerick, 2024-08-08:
> It looks like using GV's as Sources is an Edge improvement (not in OTX 2.3.15), perhaps a selectable option to upscale the GV*10.24 first, when used as a Source (where the existing code would then downscale it again) - kinda kludgy, but..??

I think so, yes... first it was just inputs, and then now it has been made available as a mixer source also, as well as for weight and offset (in the nightlies, for 2.11). Whereas I don't think you could use GVs in OTX this way at all. But whatever we do, it needs to be an off by default option, as any change would effect everyones configs, both OTX and ETX ;)

That sort of application is what I was just thinking the adjust from source and then using GV as source would be great for... flick a switch (or hold down a momentary) move the joystick where you want it, flick it back (or release)... instant set/retain position... only a couple of lines needed to configure.

DocTrucker, 2024-08-08:
Coming here from #4007!

Recently got confused by this as there is a way to set a global variable as %, and along with the special function to adust a global variable to % the mentioned result by the OP is certainly counter intuative.

philmoz, 2024-08-08:
Adding a new Adjust type looks fairly straightforward.

Looking at this, I think 'Mixer source' is a poor name for the mode - it can be any source.

I'm suggesting using 'Source (%)' and 'Source (value)' as the mode names:
[image]
[image]

pfeerick, 2024-08-08:
> I'm suggesting using 'Source (%)' and 'Source (value)' 

Sounds perfect to me... matches the label next to the source field in the Inputs/Mixes/etc screen, so is consistent that way also. :heart_eyes_cat: Should get away with no translation for that also... TR_SOURCE and TR_VALUE are both there ;)

sande005, 2024-08-08:
You guys are the best - such rapid response to items!

PR #5406 (2024-08-15): feat: new SF GV Adjust mode to copy source (instead of %) value
Fixes #5403 

- Companion strings changed to match firmware.
- STR_MIXSOURCE changed to add '(%)', language translations modified using STR_SOURCE string.
- STR_MIXSOURCERAW string added, language translations created by combining STR_SOURCE and STR_VALUE.

Language translations need to be checked.
closed 2024-08-15 by pfeerick