EdgeTX/edgetx #5411: GPS Telemetry Values missing and Decimal Point Precision Issues on Radiomaster TX12 Mk2 Internal ELRS with EdgeTX 2.10.3

johndoe422, 2024-08-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After upgrading my Radiomaster TX12 Mk2 to EdgeTX version 2.10.3 (Centurion) from 2.8.0, and ELRS firmware to 3.4.3 from 3.3.0, I encountered a significant issues where GPS telemetry values - they stopped displaying. Additionally, the telemetry screen no longer shows decimal point precision in channel values. Deleted all telemetry values and rescanned multiple times, but to no avail.

### Expected Behavior

All GPS telemetry values (speed, heading, coordinates, satellites, etc.) should be visible in the telemetry sensor list.
Decimal point precision should be maintained in the telemetry screen channel values.

Actual behavior seen: GPS telemetry values are missing. Despite a solid 3D fix and proper GPS functionality in Betaflight, the GPS-related telemetry values do not appear. Decimal point precision in channel values on the telemetry screen is absent.

Additional Information:
Downgrading from EdgeTX 2.10.3 to EdgeTX 2.9.4 (Providence) resolves the issue, restoring GPS telemetry and decimal precision. ELRS versions stayed at 3.4.3.
Packet Rate: D250 (-104dB)
Telemetry Ratio: 1:16
GPS telemetry was functioning correctly before the EdgeTX and ELRS upgrades.

Affected Hardware:

Transmitter: Radiomaster TX12 Mk2
Flight Controller: Speedybee F405 V4
Firmware: Betaflight 4.5
Receiver: GEPRC 2.4 GHz Nano

### Steps To Reproduce

Steps to Reproduce:

1) Upgrade Radiomaster TX12 Mk2 firmware from EdgeTX 2.8.0 to 2.10.3.
2) Upgrade ELRS firmware from 3.3.0 to 3.4.3 on the internal transmitter module and the GEPRC 2.4 GHz Nano receiver.
3) Verify that the GPS module on the quad (connected to Speedybee F405 V4 running Betaflight 4.5) is functioning correctly in Betaflight (GPS fix, coordinates, speed, etc. visible).
4) Check telemetry sensor list on the Radiomaster TX12 Mk2. Only 18 sensors are displayed instead of the previous 24, with GPS telemetry values missing.
5) Observe the telemetry screen; decimal point precision is missing in channel values.

### Version

2.10.3

### Transmitter

RadioMaster TX12 / TX12MK2

johndoe422, 2024-08-11:
Yes, several times, as I have mentioned in the bug description. I attempted rediscovery, deleting all and then a rediscovery, nothing helped. Only 18 sensors instead of the previous 24. 
24 sensors came back on the first scan after downgrading EdgeTx to 2.9.4 Providence.

3djc, 2024-08-11:
Well it can work since it does for me, unsure what is different in your setup so that it doesn’t

pfeerick, 2025-02-17:
> Decimal point precision should be maintained in the telemetry screen channel values.

Showing of decimals is now governed by the PPM Units runtime setting (rather than compile setting). `SYS -> RADIO SETUP -> PPM Units`. Change it from `0.--` to `0.0` if you want to have the decimal.