EdgeTX/edgetx #5421: "Centurion", v2.10.3. wrong physical joystics mappings on RadioMaster Pocket

hacky-swan, 2024-08-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The Transmitter is quite new and has never been repaired/soldered/broken.

1) Set AETR mappings in Radio Settings and in Mixers
2) Go to Inputs tab and set: 
ail - ail
ele - ele
thr - thr
rud - rud

3) Edit Thr input, select "Source", pull left joystick up (Throttle), setting will be changed to "Ele"(elevator), NOT throttle.
4) My observations: when left joystick pulled left/right "Source" is set  Ail (Aileron), not Rudder. Right joystick up/down = throttle, left/right = rudder

So, it seems that right and left joystics swapped somehow in software.

Joysticks is moving correctly when I calibrate them in hardware menu. (left one moving a dot in the left area, right one moving a dot in the right area). But there is wrong behavior in Inputs menu as I mentioned.

How to fix this?

### Expected Behavior

1) Set AETR mappings in Radio Settings and in Mixers
2) Go to Inputs tab and set: 
ail - ail
ele - ele
thr - thr
rud - rud

3) Edit Thr input, select "Source", pull left joystick up (Throttle), setting will be changed to "Thr"(Throttle), NOT elevator.

### Steps To Reproduce

1) Set AETR mappings in Radio Settings and in Mixers
2) Go to Inputs tab and set: 
ail - ail
ele - ele
thr - thr
rud - rud

3) Edit Thr input, select "Source", pull left joystick up (Throttle), setting will be changed to "Ele"(elevator), NOT throttle.
Joysticks behavior swapped.

### Version

2.10.3

### Transmitter

RadioMaster Pocket

### Operating System (OS)

Windows

### OS Version

Windows 11

### Anything else?

https://github.com/user-attachments/assets/1d31688c-b1ef-47cb-bfda-0f5b01f9ed4e

Video

philmoz, 2024-08-12:
It sounds like you changed the radio to mode 3 instead of changing the default channel order.

[image]

The little pictures above the channel names in Mode are supposed to represent the physical stick position on the radio.

hacky-swan, 2024-08-12:
> It sounds like you changed the radio to mode 3 instead of changing the default channel order.

thank you a lot! My fault. Problem was in wrong Mode.

closed 2024-08-12 by hacky-swan