EdgeTX/edgetx #5422: Jumper T15 Display low bridness noise

druckgott, 2024-08-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If the bridness of the display goes lower after some second. There comes a beeep noice 

https://github.com/user-attachments/assets/cb5c254f-b930-4b40-a95d-a352b0317856

See Video 

### Expected Behavior

No noice or switch off any where? 

### Steps To Reproduce

Let the display bridness go down due to inactivity. 

### Version

2.10.3

### Transmitter

Other (Please specify below)

### Operating System (OS)

Windows

### Anything else?

Jumper t15

pfeerick, 2024-08-12:
The whine you can hear is ether comming from the switching regulators, and/or may be noise from the RF module if the audio circuit is still active. This is a hardware limitation - nothing we can do about that. Make sure you have `Radio Settings -> Hardware -> Mute if no sound` enabled to prevent noise coming through the audio circuit when it is not actively in use.

closed 2024-08-12 by pfeerick

druckgott, 2024-08-13:
This works
Radio Settings -> Hardware -> Mute if no sound