EdgeTX/edgetx #5427: Quicker power on and not holding power button for 5 seconds

SamSpiri, 2024-08-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Holding down a power button is a time consuming process. It also requires 2 hands to do this, so i'm fully occupied by this process. I'd like to continue preparing my other stuff to fly instead of just holding the button down. Setting the startup delay in the configuration doesn't actually help. Because it is still 5 second delay, while setting is at 0.

### Describe the solution you'd like

Shorten the required time to hold down the button.

### Describe alternatives you've considered

Change the button to a switch, or add some circuit to make the button act like a switch. But it is an ugly workaround.

### Additional context

 I'd try creating a PR if someone shows me code where i could change this behavior.

pfeerick, 2024-08-13:
For what radio? There was an experimental PR to bring the MCU startup speed down on TX16S, but basically there is no artificial delay after the 2.5-3 seconds needed for the system to boot up enough to latch the power itself. The startup delay you are referring to is actually (as is labelled), the splash screen delay, for those people who want longer splash screens...

SamSpiri, 2024-08-13:
@pfeerick I got radiomaster Pocket.
The moment i push the button screen turns on, but it's blank. Then after 3 seconds it starts to draw 4 small rectangles 1 each 0.5 sec. which takes additional 2 seconds. Maybe we could shorten the box drawing procedure to 0?

pfeerick, 2024-08-13:
B&W screen radios are completely different, they up faster than colorlcd
(but there is still  the delay to actually init and read the SD card that
cannot be avoided). The "box drawing procedure" taking 2 seconds means you
still have the pwr (power) on delay  (not splash screen, that comes *after*
radio has latched the power itself - i.e.  the haptic vibration at startup
is the signal that is ok to release the power button) still set to 2
seconds! 🤣

Any other delay after setting Pwr On delay to zero is due to the SD card
performance - i.e. you could replace the microSD card with a better
performing one than the stock card.

3djc, 2024-08-15:
Tested on pocket, hold time is bellow 2 seconds on pocket and power on delay set to 0

SamSpiri, 2024-08-15:
RESOLVED: Replaced stock SDCARD and boot time is super fast!

So I just tried to copy the content to another sdcard. And now it doesn't show any 4 boxes it just starts very fast! As you said faster than 2 secs. Lol. The SD card which comes in stock with the radio from the factory is trash... Because the card I replaced it with is the SDCARD which was bought 5+ years ago 2GB.

closed 2024-08-15 by SamSpiri