EdgeTX/edgetx #5432: Screen gone crazy after latest nightly update

Kevltan, 2024-08-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Screen flickering, trims bouncing all over the place, eraneous voice calls to name just a few.

### Expected Behavior

None of above!

### Steps To Reproduce

As soon as transmitter boots into a model

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12

Kevltan, 2024-08-14:
Latest H7 SPI support seems to effect TX12S adversely.
@raphaelcoeffic
@pfeerick

3djc, 2024-08-14:
Could you please try a firmware out of this PR: https://github.com/EdgeTX/edgetx/pull/5433 ?

Kevltan, 2024-08-14:
83d663b2

pfeerick, 2024-08-14:
> TX12S does NOT like firmware main:-
3bdec2f

typo, or do you have a TX12 also? If so, this is the only radio of the three mentioned that #5433 will help with (i.e. b&w)

Kevltan, 2024-08-14:
Typo 🤣😂🤣

raphaelcoeffic, 2024-08-15:
Also worth mentioning: the X12S and its ADC is the only target to make use of SPI word (16 bit) transfers.

Kevltan, 2024-08-15:
@raphaelcoeffic 
Yes this fixes the X12S. Thanks 👍😊😊👍

PR #5436 (2024-08-15): fix: SPI word transfer
The issue was introduced in #5425.

Fixes #5432
closed 2024-08-15 by raphaelcoeffic