EdgeTX/edgetx #5434: Inverted mix switch not saving

gapster15, 2024-08-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Inverting a switch on the mixes screen will not be saved upon rebooting my MT12. 
[image]
After reboot
[image]

### Expected Behavior

Save the setting

### Steps To Reproduce

1) In mixes, select a channel using a switch
2) Set the switch to inverted and save
3) Reboot TX and the switch is no longer inverted

### Version

2.10.3

### Transmitter

RadioMaster MT12

pfeerick, 2024-08-15:
I was able to reproduce this on MT12, and on Zorro. The MT12 appeared to loose the plot one time, and change the source and drop the inverted switch completely, and the Zorro initially appear to work, but after swapping the SRC, it also lost the inverted switch on reboot.

pfeerick, 2024-08-15:
Ah, ok, I think I just spotted what is happening... using the popup menu to invert is not triggering model settings dirty... so it doesn't save... if you instead use cursor, or edit *some other* setting 'normally', it does save. ;)

philmoz, 2024-08-15:
Looks like an old issue - happens in 2.9 as well (simulator).

pfeerick, 2024-08-16:
Couldn't repo this on as easily on the the X9D+ I have that is still on 2.9.2, so maybe there was a radio navigation element to it? Either way, #5435 indeed seems to resolve this.

PR #5435 (2024-08-16): fix(bw): switch & source changes may not be saved to the SD card.
Fixes #5434 

Move check for changed value to after popup menu is processed.
closed 2024-08-28 by pfeerick