EdgeTX/edgetx #5444: In Companion 2.11.0 Simulation Mixes GV...SRC no selection

koaxheli, 2024-08-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

In Companion 2.11.0 #2292 from today, the switches GV --> SRC in the Mixes menu are not available for selection in the simulation.

### Expected Behavior

Should behave like in Companion 2.10.3.

### Steps To Reproduce

If you go to 2.10.3, go to Mixes and GV can be updated.
Go to 2.11.0 and the new switch SRC does not work.

### Version

2.10.3

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### Anything else?

[image]
[image]
[image]

pfeerick, 2024-08-16:
This appears to be a bug in the only the windows simulator... confirmed by three separate people now. Linux, and presumably MacOS (given who made the SRC/GV changes 😆) are both working as expected.

koaxheli, 2024-08-16:
Take this file under Windows and when you open the mixer in the simulation it says SRC and does not work.
The GV switches are still present for the servo outputs OK.
[image]

PR #5450 (2024-08-17): fix(simulator): SourceNumberEdit not working on Windows
Fixes #5444

For some reason the GCC version in MSYS does not compile correctly.
closed 2024-08-17 by pfeerick