EdgeTX/edgetx #5446: Channel Overrides are allowed to go past outputs

craigywsm, 2024-08-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

[test model.zip](https://github.com/user-attachments/files/16641569/test.model.zip)
With the above model I have set the low throttle to -83.0 on the outputs (1075us)
[image]

If I set a special function (sa - middle) to override channel to -100
[image]
It sets the the output to 988us
[image]

### Expected Behavior

Outputs are there to restrict servo binding e.g. total output limits
Why are special functions (override channels) allowing a channel to be over driven?

### Steps To Reproduce

1. with model attached
2. on outputs page set sa -

### Version

2.10.3

### Transmitter

FrSky X9D+2019

### Operating System (OS)

Windows

### OS Version

Windows 10 x64

pfeerick, 2024-08-16:
> Is there an existing issue for this problem?
>
>    - I have searched the existing issues
>
> What part of EdgeTX is the focus of this bug?
>
> Transmitter firmware
> Current Behavior
>
> test model.zip
> With the above model I have set the low throttle to -83.0 on the outputs
> (1075us)
> image.png (view on web)
>
> If I set a special function (sa - middle) to override channel to -100
> image.png (view on web)
> It sets the the output to 988us
> image.png (view on web)
> Expected Behavior
>
> Outputs are there to restrict servo binding e.g. total output limits
> Why are special functions (override channels) allowing a channel to be
> over driven?
> Steps To Reproduce
>
>    1. with model attached
>    2. on outputs page set sa -
>
> Version
>
> 2.10.3
> Transmitter
>
> FrSky X9D+2019
> Operating System (OS)
>
> Windows
> OS Version
>
> Windows 10 x64
> Anything else?
>
> *No response*

If I had to guess, I would guess it was in the name... It is an override...
Not a fake input, not a default. This is one of the reasons using a
additional mixer line controlled by a switch is preferred. They each have
their place. It's really up to you to configure them correctl.

3djc, 2024-08-17:
By design

craigywsm, 2024-08-17:
Fair enough

closed 2024-08-17 by craigywsm