EdgeTX/edgetx #5451: SD Logs: Value cannot be cleared

twistedwings, 2024-08-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Special Functions:
SD Logs: Value cannot be cleared (0 = off), min is 0.1

### Expected Behavior

Special Functions:
SD Logs: Value can be cleared (0 = off),

### Steps To Reproduce

1. Modell Settings
2. Special Functions
3. Create new
4. SD Logs
5. Set value to 0

### Version

2.10.3

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2024-08-17:
You can disable the SF , you can delete it, but a sd log with a 0 sec interval means nothing.

What are you trying to achieve?

twistedwings, 2024-08-17:
I want to switch logging off.
The [manual](https://manual.edgetx.org/bw-radios/model-select/special-functions) says:
"The value options are 0.0s - 25.5s (Note: 0.0 effectively disables this option)."

In companion I can set a value of 0.

3djc, 2024-08-17:
Simply disable the trigger for the sf or the sf itself

twistedwings, 2024-08-17:
that's a little bit tricky while flying...

3djc, 2024-08-17:
No, simply put the trigger for sd log on a switch or something like this

twistedwings, 2024-08-17:
I have done that.
But I want to switch logging on and off.
Therefore I have to set the parameter to 0 (for switching off)

[image]
[image]

That works in companion but not on the radio.

3djc, 2024-08-17:
This makes no sense, if L13 is not true, there will be no logs, therefore your second line is useless

twistedwings, 2024-08-17:
OK, than it was a misunderstanding on my part.
Sorry, I will close the issue.

3djc, 2024-08-17:
I think there is an issue companion (and documentation) side, so please leave open

PR #5561 (2024-09-29): fix(cpn): sf/gf sd logs minimum and default time
Fixes #5451 but not documentation!

Summary of changes:
- set minimum time to 0.1s
- set default time to 1.0s
closed 2024-09-29 by pfeerick