EdgeTX/edgetx #5454: ssue with T15 when trying to use BT mode (Frsky Modul Trainer Mode wireless)

druckgott, 2024-08-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have used this modul (CC2540 RF-BM-S02)
https://de.aliexpress.com/item/1005003109678026.html
[image]
Flasehd it like here:
https://www.instructables.com/Bluetooth-on-Radiomaster-TX16S/
And connect it.
Telemetry is working fine with  version from this pull request.

https://github.com/EdgeTX/edgetx/pull/5445

But if I select Trainer mode and go to Scann then it crashed. (see Video)

https://github.com/user-attachments/assets/aba7ec67-129b-421f-bad5-960907432120

Connection:
[image]
[image]
[image]
[image]

### Expected Behavior

No Crash ;-)

### Steps To Reproduce

Connect the modul https://de.aliexpress.com/item/1005003109678026.html
bevor flash it like here: 
https://www.instructables.com/Bluetooth-on-Radiomaster-TX16S/
Do it like here:
https://github.com/user-attachments/assets/aba7ec67-129b-421f-bad5-960907432120

### Version

2.10.3

### Transmitter

Other (Please specify below)

### Operating System (OS)

Windows

pfeerick, 2024-08-18:
Is this with the build I gave you before, or a cloudbuild using buddy (meaning this is actually 2.11)? As I'll need to give you a new for 2.10 as some stuff changed after I built that...

druckgott, 2024-08-18:
The one from. Here
https://github.com/EdgeTX/edgetx/pull/5445#issuecomment-2294504031

pfeerick, 2024-08-18:
Can you try again with this firmware. This is the latest 2.10 with bluetooth enabled. 
[t15-f6cfec59e.zip](https://github.com/user-attachments/files/16647922/t15-f6cfec59e.zip)

druckgott, 2024-08-18:
The connection now works. But I dont get a singnal if I aktivate the trainer.

druckgott, 2024-08-18:
So I did some checkt:
Trainer FrSky Taranis lite pro, student T15 --> is working --> says trainer connection
Trainer T15, student Taranis lite pro --> connection works, but scanning takes sometimes really long. And if connected, T15 says nothing and no signal if I add a switch for Trainer Axis.

3djc, 2024-08-18:
Since the same BT parsing code is run in the two radio, that points to a non functional implementation in your custom bt module

druckgott, 2024-08-18:
somethimes if I leave the page and come again, it looks like this
[image]
I have to switch to another mode and go back to see the buttons again.

3djc, 2024-08-18:
Well unsure this module can work, since T15 doesn't have an enable pin that TX16S and FrSky radios. You need a module made just for plain serial, like https://btwifimod.gitbook.io/untitled/getting-started/hardware

druckgott, 2024-08-18:
Ah ok that maybe explain it that there is no enable pin. Mhhh Then i can use a normal Modul and only telemetry. I dont what to by again more Module 😁. Because the Trainer is only nice but Not aß this important that i will by again some more Module 🤪

druckgott, 2024-08-19:
I have take a Look there is a second area for connection, the question is if this is possible to use because there is something with reset. 
[image]

3djc, 2024-08-19:
Reset and enable have nothing in common

mha1, 2024-08-19:
Both are connectors for JTAG/SWD in-circuit debugging. 4 pin: SWD, 5 pin: JTAG