EdgeTX/edgetx #5466: RXSR receiver can't be bind with telemetry using D16

JimB40, 2024-08-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Testing:
- Radios: FrSky XLite Pro, RM Pocket, RM Zorro
- External Module: FrSky XJT Lite module
- Receiver: FrSky RXSR flashed with ACCST v1.0 firmware

When binding on FrSky XLitePro and RM Pocket only
CH1-8 Telem OFF
CH9-16 Telem OFF
options are displayed

Missing
CH1-8 Telem ON
CH9-16 Telem ON

The same binding procedure on RM Zorro displays all 4 options

### Expected Behavior

Options with Telemetry ON should be displayed

### Steps To Reproduce

As in Current behavior

### Version

2.10.3

### Transmitter

FrSky X9 Lite / Lite S

3djc, 2024-08-21:
How are your module set ? (int/ext)...

pfeerick, 2024-08-21:
I can repo this on 2.10.3 on Pocket hardware, and also in Companion simulator...
Internal RF turned off, XJTlite selected for external RF, only the Telem OFF options are shown.

PR #5468 (2024-08-28): fix: xjt lite telem binding
External module FrSky telem binding should only be disabled when internal module is already using sport.

This fixes #5466
closed 2024-08-28 by pfeerick